Seeed-Studio/Linux_for_Tegra #2: Integrate with Yocto's meta-tegra

redeexpressos, 2024-05-10:
Hello. I want to build a custom image for Orin NX, using A603 carrier. However, the "tegra" Yocto layer does not have support to the A603 carrier. Is there any possible way to integrate this custom kernel to meta-tegra/linux-tegra ?

https://github.com/OE4T/meta-tegra/tree/master/conf/machine
https://github.com/OE4T/linux-tegra-5.10

Pillar1989, 2024-05-15:
@redeexpressos   Hello, there is no customized kernel for our product, we just modify the device tree and add some kernel modules, you can use our product in accordance with the official way. Of course, if you still need our Jetson products to support the Yocto framework, we can help you.

redeexpressos, 2024-05-22:
> @redeexpressos Hello, there is no customized kernel for our product, we just modify the device tree and add some kernel modules, you can use our product in accordance with the official way. Of course, if you still need our Jetson products to support the Yocto framework, we can help you.

Thank you very much for your answer. I created a new issue, regarding to my question: #4

closed 2024-05-22 by redeexpressos