Seeed-Studio/Linux_for_Tegra #3: Jetpack 6 Update

xychelsea, 2024-05-21:
The configuration files and DTB trees need to be updated to work with Jetpack 6. Specifically, I am using the ReServer Industrial Orin NX 16GB.

Pillar1989, 2024-05-22:
We are working on this and expect an update in the coming weeks. @xychelsea 
cc @bigbearishappy

bhaskara-res, 2024-06-11:
@bigbearishappy, @Pillar1989  could you please share any updates on timelines ?

bigbearishappy, 2024-06-17:
@bhaskara-res Our base work on orin nano and orin nx is finished. We are now working on different carrier board, which include recompter J401, recomputer industrial J201 and reserver industrial J401 ...

bigbearishappy, 2024-07-09:
@xychelsea @bhaskara-res 
We have uploaded the code of seeed's carrier board to the branch r36.3.0. You can now try it by following the README.

thomaskleiven, 2024-07-17:
First, thanks for your support on this topic. 

We've followed all the steps on the `r36.3.0` branch and successfully flashed our `reComputer Industrial J4012`. However, it appears that some configuration files and DTB trees are not correctly mapped. Specifically, we need `/dev/ttyTHS0` to be available, as outlined in this [NVIDIA forum issue](https://forums.developer.nvidia.com/t/orin-nano-jetpack-6-0-uart1-issue/299825).

Here's the output from `sudo dmesg | grep tty`. We would expect to see logs related to `ttyTHS0`, similar to the behavior observed in Jetpack 5.x.

```sh
sudo dmesg | grep tty
[    0.000000] Kernel command line: root=PARTUUID=3675fdfb-22bf-4708-877b-27007408e511 rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0 nospectre_bhb video=efifb:off console=tty0 bl_prof_dataptr=2031616@0x471E10000 bl_prof_ro_ptr=65536@0x471E00000 
[    0.000329] printk: console [tty0] enabled
[    0.064691] 31d0000.serial: ttyAMA0 at MMIO 0x31d0000 (irq = 117, base_baud = 0) is a SBSA
[    0.154771] printk: console [ttyTCU0] enabled
[    1.659897] printk: console [tty0]: printing thread started
[    1.660634] printk: console [ttyTCU0]: printing thread started
[    1.736486] 3100000.serial: ttyTHS1 at MMIO 0x3100000 (irq = 112, base_baud = 0) is a TEGRA_UART
[    1.737445] 3140000.serial: ttyTHS2 at MMIO 0x3140000 (irq = 113, base_baud = 0) is a TEGRA_UART
```

Is this expected or do the DTB trees need to be updated?

bigbearishappy, 2024-07-18:
> First, thanks for your support on this topic.
> 
> We've followed all the steps on the `r36.3.0` branch and successfully flashed our `reComputer Industrial J4012`. However, it appears that some configuration files and DTB trees are not correctly mapped. Specifically, we need `/dev/ttyTHS0` to be available, as outlined in this [NVIDIA forum issue](https://forums.developer.nvidia.com/t/orin-nano-jetpack-6-0-uart1-issue/299825).
> 
> Here's the output from `sudo dmesg | grep tty`. We would expect to see logs related to `ttyTHS0`, similar to the behavior observed in Jetpack 5.x.
> 
> ```shell
> sudo dmesg | grep tty
> [    0.000000] Kernel command line: root=PARTUUID=3675fdfb-22bf-4708-877b-27007408e511 rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0 nospectre_bhb video=efifb:off console=tty0 bl_prof_dataptr=2031616@0x471E10000 bl_prof_ro_ptr=65536@0x471E00000 
> [    0.000329] printk: console [tty0] enabled
> [    0.064691] 31d0000.serial: ttyAMA0 at MMIO 0x31d0000 (irq = 117, base_baud = 0) is a SBSA
> [    0.154771] printk: console [ttyTCU0] enabled
> [    1.659897] printk: console [tty0]: printing thread started
> [    1.660634] printk: console [ttyTCU0]: printing thread started
> [    1.736486] 3100000.serial: ttyTHS1 at MMIO 0x3100000 (irq = 112, base_baud = 0) is a TEGRA_UART
> [    1.737445] 3140000.serial: ttyTHS2 at MMIO 0x3140000 (irq = 113, base_baud = 0) is a TEGRA_UART
> ```
> 
> Is this expected or do the DTB trees need to be updated?

Actually, the ttyTHS0 in JP5 has been changed to ttyTHS1 in JP6, you can have a try with it~

thomaskleiven, 2024-07-18:
We are using `pin 1 (Data-)` and `pin 2 (Data+)` of the `COM 232/422/485 - DB9 J1` connector to communicate with various hardware over the `RS485` protocol, as referenced in the [Seeed Studio documentation](https://files.seeedstudio.com/products/NVIDIA/reComputer-Industrial-Reference-Guide.pdf).

Under `Jetpack 5`, communication over the `/dev/ttyTHS0` port worked very well. After upgrading to `Jetpack 6`, the `/dev/ttyTHS0` port appears to be disabled (as you mentioned). We have attempted to use the `/dev/ttyTHS1` and `/dev/ttyTHS2` ports, but we are not receiving any signals through these ports.

Is there a way to re-enable or configure the `/dev/ttyTHS0` port in `Jetpack 6`, or do you know any alternative solutions for `RS485` communication on `Jetpack 6`?

thomaskleiven, 2024-07-18:
When using the new image provided in #5, it works as expected. I'm not entirely sure what differs in this image compared to when we followed the steps on the `r36.3.0` branch, I guess we must have done something wrong there. 

As you mentioned, the memory address `0x3100000` now maps to `/dev/ttyTHS1`, whereas it mapped to `/dev/ttyTHS0` on `Jetpack 5`. So, we'll be using `/dev/ttyTHS1` going forward.

For others experiencing similar issues, I recommend using the images provided in the [Seeed documentation](https://wiki.seeedstudio.com/reComputer_Industrial_Getting_Started/#flash-to-jetson).

Thanks for the support!

bigbearishappy, 2024-07-18:
@thomaskleiven Thank you for your feedback. 
The code on branch `r36.3.0` is actually the same with the source code which generate the image you download from our wiki.
We would recommend you to use the code on `r36.3.0` if you need to add custom code for your device.

bigbearishappy, 2024-07-19:
Sine the code for JP6 is uploaded. Closing this issue. If you have any further problem. Please create a new issue~

closed 2024-07-19 by bigbearishappy