Seeed-Studio/Linux_for_Tegra #4: Confused on what board config to use (J401, Orin NX)

redeexpressos, 2024-05-22:
Hello, I've seen J401 documentation on how to flash Jetpack: https://wiki.seeedstudio.com/reComputer_J4012_Flash_Jetpack/#flash-to-jetson

for Jetpack 5.1.2 the following command its used:
```
sudo ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1 \
  -c tools/kernel_flash/flash_l4t_external.xml -p "-c bootloader/t186ref/cfg/flash_t234_qspi.xml" \
  --showlogs --network usb0 p3509-a02+p3767-0000 internal
  
```
for Jetpack 6 its

```
sudo ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1 \
  -c tools/kernel_flash/flash_l4t_t234_nvme.xml -p "-c bootloader/generic/cfg/flash_t234_qspi.xml" \
  --showlogs --network usb0 jetson-orin-nano-devkit internal
  ```
  
When building with `meta-tegra` (which uses `5.2.1`), I am confused on which board configuration should I use, `p3509-a02+p3767-0000` or `jetson-orin-nano-devkit` ?

Pillar1989, 2024-05-22:
@redeexpressos  Here you have to distinguish the meaning of the p3509, which stands for jetson-xavier-nx-devkit carrier, and the p3767, which stands for Orin series SoM. jetson-orin-nano-devkit is actually p3768-0000+p3767-0000. It represents the use of the p3768, also known as jetson-orin-nano-devkit carrier, and the p3767 is an Orin series SoM.
[image]

bigbearishappy, 2024-09-11:
Closing this issue. If you have any further problem. You can reopen it or create a new one.

closed 2024-09-11 by bigbearishappy