Seeed-Studio/Linux_for_Tegra #5: Wrong download image link for reComputer Industrial J4012

thomaskleiven, 2024-07-17:
In your [flashing guide](https://wiki.seeedstudio.com/reComputer_Industrial_Getting_Started/#flash-to-jetson) - it seems like the image download links for `reComputer Industrial J4012` and `reComputer Industrial J4011` are both pointing to the 8GB version. Can you please provide the correct link for the J4012 (16GB)? 

Thanks a lot in advance.

bigbearishappy, 2024-07-18:
I will ask help from our AE team and feedback to you.

yuyoujiang, 2024-07-18:
Hi @thomaskleiven , Thank you for your feedback. The download link has been updated. You can now [re-download](https://wiki.seeedstudio.com/reComputer_Industrial_Getting_Started/#flash-to-jetson) the required system image file.

https://github.com/Seeed-Studio/wiki-documents/pull/1414#issuecomment-2235240264

thomaskleiven, 2024-07-18:
Great - thanks. Works as expected now.

closed 2024-07-18 by thomaskleiven