Seeed-Studio/Linux_for_Tegra #6: DI/DO on J3011 not working with JP6

NiemalsSebi, 2024-07-27:
As described in the title I am not able to use the DI/DO (specifically I want to use DO0 [PI.00]). However I have tried in JP 5.1.1 and it worked without any issue using gpiod versions.

I have read online that it has to do something with the pinmux file however by Nvidias guide it should be generated using an excel file, which has all IO Pins listed. I cannot find it for the ReComputer Industrial. 

Kind regards

sniper7kills, 2024-07-28:
I'm currently looking at a similar issue trying to enable my 5G module on both the ReServer Industrial+ ReComputer Industrial.
I had opened an issue on the wiki's repo: https://github.com/Seeed-Studio/wiki-documents/issues/1348
But this is likely a better location to seek a resolution.

Regarding the excel file; I was looking at the one provided by nvidia: https://developer.nvidia.com/downloads/jetson-orin-nx-and-orin-nano-series-pinmux-config-template from https://developer.nvidia.com/embedded/downloads

But like you would love to get the one referenced in the `.dtsi` files.

bigbearishappy, 2024-08-07:
@NiemalsSebi Sorry for my late reply. I did a quick fix about this problem in this [commit](https://github.com/Seeed-Studio/Linux_for_Tegra/commit/ec222c2216bd9ac32db39e5e9af05f28f2037dc5).
You can reflash you device by following the steps in readme or download the new system image after we release it on seeed's wiki.

NiemalsSebi, 2024-08-07:
Thanks for your reply.
I have already fixed it myself, however I haven't had the time to commit a
request.

However I've got another question. do you know if it is possible to use a
m.2 key wifi card on the industrial system with an adapter?

kind regards

bigbearishappy, 2024-08-08:
@NiemalsSebi It should work fine with a adapter if your wifi card has a USB3.0 interface~

bigbearishappy, 2024-09-11:
Closing this issue. If you have any further problem. You can reopen it or create a new one.

closed 2024-09-11 by bigbearishappy