Seeed-Studio/Linux_for_Tegra #7: reComputer orin j401 fails to reboot automatically

plcarterco, 2024-08-19:
We've successfully flashed reComputer Orin NX 16gb following the repo (branch r36.3.0) instructions (thank you) and with the below command. 

sudo ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1   -c tools/kernel_flash/flash_l4t_t234_nvme.xml -p "-c bootloader/generic/cfg/flash_t234_qspi.xml" --showlogs --network usb0 recomputer-orin-j401 internal

Everything seems fine (we install jetpack, install and run deepstream, etc.) except that rebooting does not work or works the first/one time (restart from GUI) and then always seems 'stuck' in powered down, but running state, with monitor/hdmi off but does not boot back up. We disconnect and re-connect power to boot up.

Note:
-This particular device and configuration had been running for the last 12 months without issue but flashed via SDK Manager (e.g. Jetpack 6 DP in Dec 2023).
-The device has had the original 128GB SSD drive swapped out for a Samsung SSD 990 PRO 2TB (nvme) as Disk /dev/nvme0n1: 2000GB.  Note, we'd swapped in this drive  as root / boot drive for this and other Orin NXs (16gb aand 8GB) and reServer Industrial over the last 18 months and no issue flashing (with SDK Manager), running, or rebooting.
-There is a (AC8265) wifi module installed. 

It exhibits the same behavior whether reboot/restart is tested immediately after flashing or later after installing other packages. 
We reflashed as well as reformatted the disk clean and re-flashed 15+ times and all seems normal except for rebooting. 

We're not familiar with deeper internals of flashing or the various power state / management configurations and wonder if you have encountered anything similar and/or could advise.

plcarterco, 2024-08-19:
Hello,  Please disregard.  We found that there was a rootfs change made related to init.d.  Thanks

bigbearishappy, 2024-08-20:
It seems that you have resolved this problem. So closing it now~

closed 2024-08-20 by bigbearishappy