Seeed-Studio/Linux_for_Tegra #8: Instructions for yocto

CoLorenzo, 2024-09-11:
I read in the readme that you can customize the image using yocto and make it work on the boards. 
how can i create an image for yocto working for example for reComputer Industrial J4012?
do i need to use any specific layers?
thank you very much

bigbearishappy, 2024-09-11:
You don't need to use specific layers. 
And I think You can make it based on the official yocto layer of jetson.

Here are the steps for your reference:
1 add the dts file for reComputer Industrial J4012 to yocto project and enable them
2 check the kernel config to make the device driver for reComputer Industrial J4012 can work

bigbearishappy, 2024-10-15:
Closing this issue. If you have any further problem. Please reopen it or create a new one.

closed 2024-10-15 by bigbearishappy