Seeed-Studio/Linux_for_Tegra #9: Updates needed for JP6 branch Intall README

Ktysai, 2024-09-13:
Hello guys!

As a noob in jetson and  stuff there were some problems with the docs.

1. users might need 'flex' and 'bison' 
2. At step 4 , it has to be `cd ..`  and again `cd ..`
3. at step 4 the `mkdir github/Linux_for_Tegra` should me `mkdir -p github/Linux_for_Tegra`
4. for Adding user and pass since there is no HDMI output, it has to use host computer to access ttyACM0 with 115200 baudrate.

Hope to be useful for new users :)

bigbearishappy, 2024-09-14:
Thank you for your work to point the mistake. It would be nice if you can create a PR to this repo.

bigbearishappy, 2024-09-14:
I just update the code based on your work in this [commit](https://github.com/Seeed-Studio/Linux_for_Tegra/commit/e023fa6614018463894a224cbf26845d48af4ca4).

It fix the problem 2 and 3 in your list.
For problem 1. You can install it by add `apt install ...` in `extra_scripts/rootfs_magic.sh`
And we already enable the user to create account via ttyACM0 in our dts code.

Ktysai, 2024-09-14:
Thank you! Github is still new for me :)
What do you mean by:
> And we already enable the user to create account via ttyACM0 in our dts code.

Ktysai, 2024-09-16:
> I just update the code based on your work in this [commit](https://github.com/Seeed-Studio/Linux_for_Tegra/commit/e023fa6614018463894a224cbf26845d48af4ca4).
>
> It fix the problem 2 and 3 in your list. [...]

If I may, those extra scripts when are run?

bigbearishappy, 2024-09-18:
> Thank you! Github is still new for me :) What do you mean by:
> 
> > And we already enable the user to create account via ttyACM0 in our dts code.

I mean that you should be able to create a new user with the USB which you used to flash you device. [Here](https://github.com/Seeed-Studio/Linux_for_Tegra/blob/e023fa6614018463894a224cbf26845d48af4ca4/source/hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j201-p3768-0000%2Bp3767-recomputer-indu-common.dts#L86) is the device tree details.

bigbearishappy, 2024-09-18:
> If I may, those extra scripts when are run?

The usage of this script is showed in the .gitlab-ci.yml. Here is the detail [link](https://github.com/Seeed-Studio/Linux_for_Tegra/blob/e023fa6614018463894a224cbf26845d48af4ca4/.gitlab-ci.yml#L90).

bigbearishappy, 2024-10-15:
Closing this issue. If you have any further problem. Please reopen it or create a new one.

closed 2024-10-15 by bigbearishappy