Seeed-Studio/Linux_for_Tegra #10: installation hangs on reComputer Industrial J4012 after flashing

CoLorenzo, 2024-09-16:
I'm facing an issue with installation on a reComputer Industrial J4012 board.
I've flashed the board using the following command:

`sudo ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1   -c tools/kernel_flash/flash_l4t_t234_nvme.xml -p "-c bootloader/generic/cfg/flash_t234_qspi.xml"   --showlogs --network usb0 recomputer-industrial-orin-j201 internal`

The flashing process completes successfully, but when I start the installation and configure the initial settings, the installation screen suddenly disappears, and the system becomes unresponsive.

If I force-quit the installation wizard using kill command, the system boots, but I encounter permission issues. I'm unable to perform any actions that require root privileges.

[image]

How can I resolve the permission issues that are preventing me from executing commands with root privileges?

What is the default password for the root user in this setup?

bigbearishappy, 2024-09-18:
Actually, I don't know the password of root user too.
Maybe you can try to create the new user with the USB port which you used to flash your device.

bigbearishappy, 2024-10-15:
Have you tried to download the system image provided in seeed's wiki?

CoLorenzo, 2024-10-15:
I tried, unfortunately with that [image](https://szseeedstudio-my.sharepoint.cn/personal/youjiang_yu_szseeedstudio_partner_onmschina_cn/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fyoujiang%5Fyu%5Fszseeedstudio%5Fpartner%5Fonmschina%5Fcn%2FDocuments%2FJetson%2DImage%2FreComputer%2Dindustrial%2Fjp6%2E0%2Fmfi%5Frecomputer%2Dindustrial%2Dorin%2Dnx%2D16g%2Dj201%2D6%2E0%2D36%2E3%2E0%2D2024%2D08%2D08%2Etar%2Egz&parent=%2Fpersonal%2Fyoujiang%5Fyu%5Fszseeedstudio%5Fpartner%5Fonmschina%5Fcn%2FDocuments%2FJetson%2DImage%2FreComputer%2Dindustrial%2Fjp6%2E0&ga=1) I get an error at the end of the flashing procedure 

[flash_tail.txt](https://github.com/user-attachments/files/17376527/flash_tail.txt)

closed 2024-10-30 by Lesords