Seeed-Studio/Linux_for_Tegra #11: activate em12-g modem gpio

Robert-Petrosian, 2024-09-17:
# I am encountering a series of issues during boot, and I’m specifically seeing the following error logs related to GPIO and IRQ mapping:
[    3.591292] gpio-394 (camera-control-output-low): hogged as output/low
[    3.608438] i2c-mux-gpio bus@0:cam_i2cmux: 2 port mux on 3180000.i2c adapter
[    3.650596] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[    9.819194] irq: type mismatch, failed to map hwirq-131 for gpio@2200000!

# Previously, I activated the modem by exporting and controlling GPIO pins manually using the following sequence of commands:
```
sudo su 
cd /sys/class/gpio
echo 309 > export 
cd gpio309
echo out > direction
echo 0 > value

cd ..
echo 341 > export 
cd PEE.02
echo out > direction
echo 1 > value

cd ..
echo 330 > export 
cd PCC.02
echo out > direction
echo 0 > value
``` 
Current Issue:

# Now, after moving away from the sysfs method (since it is deprecated), I am trying to activate the modem using the gpioset command with the following approach:
```sudo gpioset gpiochip1 25=1
sudo gpioset gpiochip1 14=0```
However, this method does not seem to activate the modem as expected. I am unsure if I am using the correct GPIO pins or if there is a problem with how I am setting the values using gpioset.

Linux quf 5.15.136-tegra #1 SMP PREEMPT Mon Sep 16 11:43:00 +05 2024 aarch64 aarch64 aarch64 GNU/Linux 
reComputer Industrial J401

```gpiochip0 - 164 lines:
        line   0:      "PA.00" "regulator-vdd-3v3-sd" output active-high [used]
        line   1:      "PA.01"       unused   input  active-high 
        line   2:      "PA.02"       unused   input  active-high 
        line   3:      "PA.03"       unused   input  active-high 
        line   4:      "PA.04"       unused   input  active-high 
        line   5:      "PA.05"       unused   input  active-high 
        line   6:      "PA.06"       unused   input  active-high 
        line   7:      "PA.07"       unused   input  active-high 
        line   8:      "PB.00"       unused   input  active-high 
        line   9:      "PC.00"       unused   input  active-high 
        line  10:      "PC.01"       unused   input  active-high 
        line  11:      "PC.02"       unused   input  active-high 
        line  12:      "PC.03"       unused   input  active-high 
        line  13:      "PC.04"       unused   input  active-high 
        line  14:      "PC.05"       unused   input  active-high 
        line  15:      "PC.06"       unused   input  active-high 
        line  16:      "PC.07"       unused   input  active-high 
        line  17:      "PD.00"       unused   input  active-high 
        line  18:      "PD.01"       unused   input  active-high 
        line  19:      "PD.02"       unused   input  active-high 
        line  20:      "PD.03"       unused   input  active-high 
        line  21:      "PE.00"       unused   input  active-high 
        line  22:      "PE.01"       unused   input  active-high 
        line  23:      "PE.02"       unused   input  active-high 
        line  24:      "PE.03"       unused   input  active-high 
        line  25:      "PE.04"       unused   input  active-high 
        line  26:      "PE.05"       unused   input  active-high 
        line  27:      "PE.06"       unused   input  active-high 
        line  28:      "PE.07"       unused   input  active-high 
        line  29:      "PF.00"       unused   input  active-high 
        line  30:      "PF.01"       unused   input  active-high 
        line  31:      "PF.02"       unused   input  active-high 
[... 141 lines cut ...]
        line   8:     "PBB.00"       unused   input  active-high 
        line   9:     "PBB.01"       unused   input  active-high 
        line  10:     "PBB.02"       unused   input  active-high 
        line  11:     "PBB.03"       unused  output  active-high 
        line  12:     "PCC.00"       unused  output  active-high 
        line  13:     "PCC.01"       unused  output  active-high 
        line  14:     "PCC.02"       unused  output  active-high 
        line  15:     "PCC.03"        "mux"  output  active-high [used]
        line  16:     "PCC.04"       unused   input  active-high 
        line  17:     "PCC.05"       unused   input  active-high 
        line  18:     "PCC.06"       unused   input  active-high 
        line  19:     "PCC.07"       unused   input  active-high 
        line  20:     "PDD.00"       unused   input  active-high 
        line  21:     "PDD.01"       unused   input  active-high 
        line  22:     "PDD.02"       unused   input  active-high 
        line  23:     "PEE.00"       unused   input  active-high 
        line  24:     "PEE.01"       unused   input  active-high 
        line  25:     "PEE.02"       unused   input  active-high 
        line  26:     "PEE.03"       unused   input  active-high 
        line  27:     "PEE.04"      "Power"   input   active-low [used]
        line  28:     "PEE.05"       unused   input  active-high 
        line  29:     "PEE.06"       unused   input  active-high 
        line  30:     "PEE.07"       unused   input  active-high 
        line  31:     "PGG.00"       unused   input  active-high 
```

MartyLee42, 2024-09-17:
The same issue too! Can't find out what i am missing

bigbearishappy, 2024-09-18:
Can you tell me the device name of yours?

Robert-Petrosian, 2024-09-18:
> Can you tell me the device name of yours?

reComputer Industrial J4011

bigbearishappy, 2024-09-19:
Please try the following command:
```
sudo gpioset --mode=wait gpiochip1 25=1
sudo gpioset --mode=wait gpiochip1 14=0
```

closed 2024-09-19 by Robert-Petrosian