Seeed-Studio/Linux_for_Tegra #14: dts file for orin-nano-industrial-4g - reComputer Industrial J3010

noob502, 2024-11-06:
Hello, I am building a custom image using Yocto build. As per the instructions in this [issue](https://github.com/Seeed-Studio/Linux_for_Tegra/issues/8), I am looking for dts/dtb file for orin-nano-industrial-4g - reComputer Industrial J3010. 

However, I did not find any device-tree file for this model. Any idea how to work? 

Thanks.

noob502, 2024-11-06:
I am working on mickledore branch which is based on Jetson Linux release: R35.4.1 JetPack release: 5.1.2, and I am using vision component camera IMX250.

noob502, 2024-11-06:
I already built an image with meta-tegra (I did not include any dts/dtb file from this repo), but neither camera is working. However, I can see the vc_mipi driver in the image.

bigbearishappy, 2024-11-26:
Actually, the hardware design of camera from seeed's reComputer Industrial J3010 is same with nvidia development kit board. We don't change the dts code of it.
As you said that you can see the vc_mipi driver in image. Have you check the device tree of you target system?
Maybe the camera is not enabled in device tree.

closed 2024-12-16 by Lesords