Seeed-Studio/Linux_for_Tegra #16: reComputer Industrial J3011 POE Camera on PSE Port

almarion, 2024-11-23:
The PSE port will work successfully with any non-POE device and with any cable configuration.

2 sample industry standard (IEEE 802.3af) POE IP cameras do NOT work with the PSE port.
Both cameras work with an industry standard (IEEE 802.3af) POE injector.

The PSE port does negotiate with the cameras and provides 50 Volts on the appropriate cable pins.  (Also ping appears to be successful.)  Compared with POE injector however, the 50 Volt polarity appears to be reversed.

Using Jetpack 6:  mfi_recomputer-industrial-orin-nano-8g-j201-6.0-36.3.0-2024-08-08.tar.gz

My conclusion is that the PSE on the J3011 is NOT providing industry standard(IEEE 802.3af) handshaking/negotiation to POE devices.  Therefore, it would seem there is a design problem and/or bug.  Perhaps a driver issue?

bigbearishappy, 2024-11-27:
Can you show me the hardware connection of your devices?

almarion, 2024-11-28:
I simply take the cable that was successfully used to connect the camera to the POE injector and connect it instead to the PSE port.

The following link shows the cable pinout.  
IEEE 802.3af standard POE pinout: T568B color
This cable was also recommended by Seeed Tech Support.

https://pinoutguide.com/Net/poe_pinout.shtml

bigbearishappy, 2024-11-29:
OK, We will try to reproduce your problem and feedback to you~

bigbearishappy, 2024-12-10:
Sorry, I can't reproduce your problem.
And I asked our hardware engineer for help about this problem. They told me the Ethernet PSE power supply module on J3011 is compliant with IEEE802.3at, which is compatible with IEEE 802.3 af.

There is another way you can have a try by disable the EEE on PSE port.
Here is the command:
```
sudo ethtool --set-eee eth0 eee off
```

almarion, 2024-12-18:
Still no success  ....eee not supported ?

alan@orin-nano-jp6:~$ sudo ethtool --set-eee eth0 eee off
alan@orin-nano-jp6:~$ sudo ethtool --set-eee eth0 eee on
netlink error: Protocol not supported
alan@orin-nano-jp6:~$ 

alan@orin-nano-jp6:~$ ethtool -i eth0
driver: lan743x
version: 5.15.136-tegra
firmware-version: 
expansion-rom-version: 
bus-info: 0001:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: yes
supports-register-dump: no
supports-priv-flags: yes
alan@orin-nano-jp6:~$ 

alan@orin-nano-jp6:~$ sudo ethtool -k eth0
Features for eth0:
rx-checksumming: off [fixed]
tx-checksumming: on
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: on
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: on
	tx-scatter-gather: on
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: on
	tx-tcp-segmentation: on
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off
	tx-tcp6-segmentation: off [fixed]
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-tunnel-remcsum-segmentation: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
tx-udp-segmentation: off [fixed]
tx-gso-list: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
tls-hw-tx-offload: off [fixed]
tls-hw-rx-offload: off [fixed]
rx-gro-hw: off [fixed]
tls-hw-record: off [fixed]
rx-gro-list: off
macsec-hw-offload: off [fixed]
rx-udp-gro-forwarding: off
hsr-tag-ins-offload: off [fixed]
hsr-tag-rm-offload: off [fixed]
hsr-fwd-offload: off [fixed]
hsr-dup-offload: off [fixed]
alan@orin-nano-jp6:~$

closed 2025-01-17 by github-actions[bot]