Seeed-Studio/Linux_for_Tegra #18: ETA for 36.4 support?

gabriel-st-angel-aerospace, 2024-12-05:
Hi,

Is there a formal ETA that Seeed can provide for 36.4 support?  I can't seem to find any official information on this topic.

Thank you for your continued work!

bigbearishappy, 2024-12-06:
We are working on 36.4.0 now. And we will release it after the test work is finished.

gstangel, 2024-12-31:
@bigbearishappy Happy New Years!

I see  JP 6.1 Precompiled is now available here https://wiki.seeedstudio.com/reComputer_J4012_Flash_Jetpack/ 
Awesome, thanks.  Is the source getting merged into this repository soon?

thx

bigbearishappy, 2025-01-02:
Happy New Years!
I will sync the code for seeed's jetson board today.

bigbearishappy, 2025-01-02:
The code for R36.4.0 is uploaded now. You can get it in branch r36.4.0.

closed 2025-01-02 by bigbearishappy

gstangel, 2025-01-02:
Happy New Year!  Thank you for pushing this!!