Seeed-Studio/Linux_for_Tegra #20: Unable to build Seeed device trees from source.  File Missing

gstangel, 2025-01-02:
Hi there, thank you for releasing the 36.4 BSP patch.  There is one file missing that is needed to build the device trees from source 
**tegra234-j401-p3768-0000+p3767-recomputer-common.dts**

```
mkdir -p /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/ ; gcc -E -Wp,-MMD,/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.d.pre.tmp -nostdinc -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/tegra/nv-public -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/kernel -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/nvidia-oot -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/platforms -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public -I/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel//kernel-jammy-src/scripts/dtc/include-prefixes  -DLINUX_VERSION=600 -DTEGRA_HOST1X_DT_VERSION=2 -undef -D__DTS__ -x assembler-with-cpp -o /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.dts.tmp /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/../..//hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j401-p3768-0000+p3767-0000-recomputer.dts ; /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel//kernel-jammy-src/scripts/dtc/dtc -o /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb -b 0 -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/../..//hardware/nvidia/t23x/nv-public/nv-platform/ -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/tegra/nv-public -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/kernel -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/nvidia-oot -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public/include/platforms -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/hardware/nvidia/t23x/nv-public -i/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel//kernel-jammy-src/scripts/dtc/include-prefixes -Wno-interrupt_provider -Wno-unit_address_vs_reg -Wno-unit_address_format -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-simple_bus_reg -Wno-unique_unit_address   -@ -@ -@ -@ -Wno-interrupt_provider -Wno-unit_address_vs_reg -Wno-unit_address_format -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-simple_bus_reg -Wno-unique_unit_address   -d /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.d.dtc.tmp /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.dts.tmp ; cat /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.d.pre.tmp /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.d.dtc.tmp > /osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/.tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb.d
/osl/orin-build/active_builds/orin_nx_8gb_j401/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/../..//hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j401-p3768-0000+p3767-0000-recomputer.dts:6:10: fatal error: tegra234-j401-p3768-0000+p3767-recomputer-common.dts: No such file or directory
6 | #include "tegra234-j401-p3768-0000+p3767-recomputer-common.dts"

```

gstangel, 2025-01-02:
@bigbearishappy seems to be some .dts missing for J401 in kernel sources

bigbearishappy, 2025-01-03:
Oh, sorry for my mistake. I have uploaded the missing file just now. Please pull the latest code and try it again.

closed 2025-01-25 by github-actions[bot]