Seeed-Studio/Linux_for_Tegra #21: Real-time kernel support

nvasudevan, 2025-01-05:
Hello,

How do I enable real-time kernel support using the instructions provided in the [README](https://github.com/Seeed-Studio/Linux_for_Tegra/tree/r36.4.0?tab=readme-ov-file#linux_for_tegra)? The only set of instructions I have for R36.4 is from the [NVIDIA jetson](https://docs.nvidia.com/jetson/archives/r36.4/DeveloperGuide/SD/Kernel/KernelCustomization.html#using-the-jetson-linux-real-time-kernel-package) page but I am aware that many of the drivers is customised for a specific board and so using NVIDIA instructions may not work.

My carrier board is `Jetson Orin Nano reComputer J4012`, and currently the [instructions](https://wiki.seeedstudio.com/reComputer_J4012_Flash_Jetpack/) work fine but without the Real-time kernel support.

Are you able to point me in the right direction?

Thanks
Krish.

bigbearishappy, 2025-01-07:
It seems that you should recompile the kernel and install it to your device.
Here is the [link](https://docs.nvidia.com/jetson/archives/r36.4/DeveloperGuide/SD/Kernel/KernelCustomization.html#building-the-jetson-linux-kernel) I found in NVIDIA jetson website.
I hope this can help you.

nvasudevan, 2025-01-08:
ok, thanks. I will try the instructions from the NVIDIA site. Either way, I will post my findings here. Cheers.

closed 2025-01-31 by github-actions[bot]