Seeed-Studio/Linux_for_Tegra #23: Jetson Orin Nano 8GB - image based OTA via golden image 5.1.3 -> 6.1

deadmerc, 2025-01-20:
Hello, 

I am trying to make OTA package to update my device 5.1.3 -> 6.1
flash are working fine

`tools/ota_tools/version_upgrade/ota_board_specs.conf` was modified with missing line for support OTA

```
jetson_orin_nano_devkit_ota_emmc_r35_r36_spec=(
        ...
        # new orin-nano 8GB seeed
        'boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=recomputer-orin-j401;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
)
```
But during OTA package generation script stack with error

```
./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s --external-device nvme0n1 -S 118GiB recomputer-orin-j401 R35-5

Target board(RECOMPUTER_ORIN_J401) and base version(R35_5) is not supported
Failed to get the chip id for RECOMPUTER_ORIN_J401
OTA with nvme0n1 is not supported on recomputer-orin-j401
Usage: sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh [options] <target board> <bsp version>
...
```
Any advice are welcome

deadmerc, 2025-01-20:
File `tools/ota_tools/version_upgrade/ota_board_specs.conf`

Modified as

```
T23X_DEVICES=(
        'JETSON_AGX_ORIN_DEVKIT'
        'JETSON_AGX_ORIN_DEVKIT_INDUSTRIAL'
        'JETSON_ORIN_NANO_DEVKIT'
        'RECOMPUTER_ORIN_J401'
)
```

File `l4t_generate_ota_package.sh`
Modified as
```
SUPPORTED_EXTERNAL_DEVICES=(
        'jetson-agx-orin-devkit:nvme0n1'
        'jetson-agx-orin-devkit-industrial:nvme0n1'
        'jetson-orin-nano-devkit:nvme0n1'
        'recomputer-orin-j401:nvme0n1'
)
```

Now OTA script starts, but stuck in a loop with error

```bash
./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s --external-device nvme0n1 -S 118GiB recomputer-orin-j401 R35-5
```

```
...
[   6.1185 ] tegrahost_v2 --chip 0x23 0 --align mem_rcm_3_aligned.bct
[   6.1206 ] tegrahost_v2 --chip 0x23 0 --magicid MEM3 --ratchet_blob ratchet_blob.bin --appendsigheader mem_rcm_3_aligned.bct zerosbk
[   6.1216 ] adding BCH for mem_rcm_3_aligned.bct
[   6.1255 ] tegrasign_v3.py --key None --list mem_rcm_3_aligned_sigheader.bct_list.xml --pubkeyhash pub_key.key --sha sha512
[   6.1258 ] Assuming zero filled SBK key
[   6.1279 ] Warning: pub_key.key is not found
[   6.1284 ] tegrahost_v2 --chip 0x23 0 --updatesigheader mem_rcm_3_aligned_sigheader.bct.encrypt mem_rcm_3_aligned_sigheader.bct.hash zerosbk
[   6.1302 ] Copying signatures
[   6.1313 ] tegrahost_v2 --chip 0x23 0 --partitionlayout flash.xml.bin --updatesig images_list_signed.xml
[   6.1334 ] tegraparser_v2 --generategpt --pt flash.xml.bin
[   6.1344 ] End sector for APP, expected at: 199999966, actual: 0
Error: Return value 4
Command tegraparser_v2 --generategpt --pt flash.xml.bin
cp: cannot stat 'signed/*': No such file or directory
cp: cannot stat 'signed/flash.xml.tmp': No such file or directory
Using bpmp-dtb concatenated with odmdata in blob for t23x
XPath set is empty
...
```

bigbearishappy, 2025-01-22:
I don't have experience on upgrading from 5.1.3 to 6.1. But I have tested the OTA from 5.1.3 to 6.0 by following this [doc](https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/SD/SoftwarePackagesAndTheUpdateMechanism.html#updating-jetson-linux-with-image-based-over-the-air-update) from nvidia.

And I don't change the file like `tools/ota_tools/version_upgrade/ota_board_specs.conf` when I OTA from 5.1.3 to 6.0.

Maybe you can have a try to upgrade fro 5.1.3 to 6.0 and then from 6.0 to 6.1.

deadmerc, 2025-01-22:
HI @bigbearishappy ,

Thank you for answer, by any chance you remember which config you are used? 
 - jetson-orin-nano-devkit
 - p3509-a02+p3767-0000 (alias?)
 - recomputer-orin-j401

Any mofidication was done to layout?
 - device type="external" instance="0" sector_size="512" num_sectors="200000000"

Which image was used for OTA?
 - base system.img which can be created by first flash
 - golden image copied form second jetson

If u have commands in history, thats would be great, thank you for help

bigbearishappy, 2025-01-22:
Which config?
jetson-orin-nano-devkit

Any modifidication?
num_sectors="200000000"

Which image was used for OTA?
Sorry, I don't remember this.

deadmerc, 2025-01-22:
Thank you, I will try with this configuration

deadmerc, 2025-01-22:
@bigbearishappy could you please help me to understand how properly set value `num_sectors`

if I have ssd with 120GB total, value 200000000 come as `512*200000000 bytes as GiB = 95.36 GIB` which is not fitting to ssd or partition size, how to properly calculate this value?

also we have a second value in command it self `-S 118GiB` thats should be size of my main partition?

bigbearishappy, 2025-01-23:
I think you can flash the system with the default num_sectors and then make a disk expansion after the system booted up.
And then you can get the maxmium value of you SSD size in current system.
Assuming it's X. Then the num_sectors = X / 512.
That should be the exact value of you SSD.

deadmerc, 2025-01-23:
Hi @bigbearishappy did you experience this problem during OTA trigger script?

`The board name in OTA package(jetson-orin-nano-devkit) does not match current board(p3509-a02+p3767-0000)`

I have jetson orin nano 8gb

deadmerc, 2025-01-29:
HI @bigbearishappy if u have time, could you please check logs, I managed to make OTA and triggered it, but device are no longer responding after with error

```
L4TLauncher: Attempting Direct Boot
EFI stub: Booting Linux Kernel...
EFI stub: ERROR: Invalid header detected on UEFI supplied FDT, ignoring ...
EFI stub: Generating empty DTB
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services...
```

[jetson_minicom.txt](https://github.com/user-attachments/files/18583670/jetson_minicom.txt)

deadmerc, 2025-02-04:
Problem still relevant, up

bigbearishappy, 2025-02-05:
> Hi [@bigbearishappy](https://github.com/bigbearishappy) did you experience this problem during OTA trigger script?
> 
> `The board name in OTA package(jetson-orin-nano-devkit) does not match current board(p3509-a02+p3767-0000)`
> 
> I have jetson orin nano 8gb

Sorry, I don't.

bigbearishappy, 2025-02-05:
> HI [@bigbearishappy](https://github.com/bigbearishappy) if u have time, could you please check logs, I managed to make OTA and triggered it, but device are no longer responding after with error
>
> ``` [...]

According to the log you provided. It seems the main device tree of you device is in something wrong.

bigbearishappy, 2025-02-05:
In my experience. You should check if the main dtb file is still in you SSD's /boot/ directiory.
I am trying to reproduce you problem. And I will give you more advice if I can go further.

deadmerc, 2025-02-05:
Understood, thanks a lot

More information if it will help:

I dont remember how exactly was made device 5.1.3 as guide in Seeed wiki contain only 5.1.2, I believe was flashed 5.1.2 with `p3509-a02+p3767-0000` config, then updated to 5.1.3 via apt.

jetpack 6 was flashed by this repo with `recomputer-orin-j401` config

golden image made via these commands on J6 device:

```bash
./tools/kernel_flash/l4t_initrd_flash.sh --initrd  jetson-orin-nano-devkit nvme0n1
ssh root@fc00:1:1:0::2 "dd if=/dev/nvme0n1p1 | gzip" | gzip -d > bootloader/system.img.raw

ls -lah bootloader/system.img.raw
-rw-r--r-- 1 root root 118G Jan 16 12:15 bootloader/system.img.raw
```

num_sectors="251658240"

ota package generate command
```
./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s --external-device nvme0n1 -S 118GiB jetson-orin-nano-devkit R35-5
```

deadmerc, 2025-02-05:
Also how I understood in J6, DTB file can be inside [UEFI directly](https://docs.nvidia.com/jetson/archives/r36.4.3/DeveloperGuide/SD/Bootloader/UEFI.html#dtb-support)
J6 doesnt have a line in extlinux pointing into dtb 
```
/boot/dtb/kernel_tegra234-j401-p3768-0000+p3767-0003-recomputer.dtb

$ cat /boot/extlinux/extlinux.conf
TIMEOUT 30
DEFAULT primary

MENU TITLE L4T boot options

LABEL primary
      MENU LABEL primary kernel
      LINUX /boot/Image
      INITRD /boot/initrd
      APPEND ${cbootargs} root=PARTUUID=29e82608-5415-4a34-b0cc-96a3d0151c1b rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 nospectre_bhb video=efifb:off console=tty0
```
While J5 are 
```
/boot/dtb/kernel_tegra234-p3767-0003-p3509-a02.dtb

cat /boot/extlinux/extlinux.conf
TIMEOUT 30
DEFAULT primary

MENU TITLE L4T boot options

LABEL primary
      MENU LABEL primary kernel
      LINUX /boot/Image
      FDT /boot/dtb/kernel_tegra234-p3767-0003-p3509-a02.dtb
      INITRD /boot/initrd
      APPEND ${cbootargs} root=PARTUUID=d3580ebd-6ca0-4331-b02e-f54ba38ef076 rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 console=ttyAMA0,115200 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0 nospectre_bhb
```

after OTA I still can see my old custom UEFI `Jetson UEFI firmware (version 202210.4-a5ac12d7-dirty built on 2024-02-27T13:00: 14+00:00)` means it was not updated for some reason

bigbearishappy, 2025-02-05:
It seems that you used a another way to achieve the OTA, which is different from what I mentioned before. Can you send me the reference info where you learn this way from?

deadmerc, 2025-02-05:
Sure, the only difference from online [docs](https://docs.nvidia.com/jetson/archives/r36.4/DeveloperGuide/SD/SoftwarePackagesAndTheUpdateMechanism.html#steps-performed-on-the-host-machine) I dont generate system image, but using my one from already upgraded device

inside file `tools/ota_tools/version_upgrade/Image_based_OTA_Examples.txt` full info

```
4. Using golden image in OTA update.
Customers might flash a device with the latest released BSP and then make a golden image
dependent on this device. They can deploy the golden image on other devices that were
flashed with the previously released BSP through an OTA update.

The "l4t_generate_ota_package.sh" script includes the "-f" and "-o" options so that
customers can use their own rootfs updater and update the rootfs partition with their
own rootfs image (for example, a golden image).

However, for some customers, the golden image is a raw partition image that is
directly obtained from rootfs partition. (These customers might not
want to write their own rootfs updater.)

The following examples provide the steps for using this golden image in an image-based OTA.

Case 12: rootfs A/B disabled, disk encryption disabled:
  a. Put the target device into recovery mode.
  b. Boot the device to initrd by running the following command:
     $ sudo ./tools/kernel_flash/l4t_initrd_flash.sh --initrd <target board> mmcblk0p1
  c. Enter bash shell on the target device.
  d. Mount a USB drive or network storage on the target device.
  e. Generate golden image in the mounted storage device by running the
     following command:
     $ sudo dd if=<rootfs partition> of=<mount ponint>/golden.img.raw
     The "<rootfs partition>" is /dev/mmcblk0p1 for the internal eMMC storage device
     or /dev/nvme0n1p1 for the external NVMe storage device.
  f. Copy the golden image into the TARGET_BSP directory on the host machine by running
     the following command:
     $ sudo cp golden.img.raw ${TARGET_BSP}/bootloader/system.img.raw
  g. Add the "-s" option when generating the OTA payload package.
     For the internal eMMC storage device, running the following command:
     $ sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh \
       -s <target board> <base version>
     For the external NVMe storage device, running the following command:
     $ sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh \
       -s --external-device nvme0n1 -S <rootfs size > <target board> <base version>
```

deadmerc, 2025-02-06:
Unfortunately I dont remember exactly how it was made, as it was more than year ago, I am checking device
```
~# cat /etc/nv_tegra_release
# R35 (release), REVISION: 5.0, GCID: 35550185, BOARD: t186ref, EABI: aarch64, DATE: Tue Feb 20 04:46:31 UTC 2024

~#  apt-cache show nvidia-jetpack
Package: nvidia-jetpack
Version: 5.1.3-b29
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 5.1.3-b29), nvidia-jetpack-dev (= 5.1.3-b29)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.1.3-b29_arm64.deb
Size: 29308
SHA256: b8b35bd12e6bf07c9fe61563babf08e802db09bd8fe4611bd4c53c21daaf0b48
SHA1: b713997204bb7353f9d6edf499af68c944d0108d
MD5sum: 3199c10880f3f756308f59ffacd6a060
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8
```
probably was used official BSP from nvidia 5.1.3

deadmerc, 2025-02-06:
I checked if device was updated via apt in history
`/var/log/apt/history.log`

```
Start-Date: 2024-02-27  13:27:24
Commandline: apt install nvidia-jetpack
Requested-By: 
Install: libnpp-dev-11-4:arm64 (11.4.0.287-1, automatic), nvidia-cudnn8:arm64 (5.1.3-b29, automatic),
....
```

Thats was the only one jetpack install command, so I believe it was not updated after

bigbearishappy, 2025-02-06:
When I run `sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s --external-device nvme0n1 -S 118GiB jetson-orin-nano-devkit R35-5` to generate OTA package.
It output this error log:`Error: "BASE_BSP" is not set`.
How do you go through this problem?

bigbearishappy, 2025-02-07:
I tried to reproduce you problem but I stuck in the step of generating golden image from target device, cause the golden is so big and it takes too much time to do this.
To simply the complexity of reproduction. I decide to do a OTA upgrade without golden image from 5.1.3 to 6.1.
However, I encountered with the problem of generating ota payload package with command `sudo BASE_BSP=/path/to/jp5/Linux_for_Tegra/ ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s --external-device nvme0n1 -S 118GiB jetson-orin-nano-devkit R35-5`.

That 's so upset.
So I give another try with OTA from 5.1.3 to 6.0, and finally it works.
Here is my detail steps about how I achieve this, I hope this can help you to solve the current problem:
## 1 Generate the system image of JP5.1.3
### 1.1 Prepare the SDK
```
cd jp5
wget https://developer.nvidia.com/downloads/embedded/l4t/r35_release_v5.0/release/jetson_linux_r35.5.0_aarch64.tbz2
wget https://developer.nvidia.com/downloads/embedded/l4t/r35_release_v5.0/release/tegra_linux_sample-root-filesystem_r35.5.0_aarch64.tbz2
```
### 1.2 Build system image of JP5
```
tar xf jetson_linux_r35.5.0_aarch64.tbz2
sudo tar xpf tegra_linux_sample-root-filesystem_r35.5.0_aarch64.tbz2 -C Linux_for_Tegra/rootfs/
sudo ./apply_binaries.sh
sudo  BOARDID=3767 BOARDSKU=0001 FAB=300  BOARDREV=G.0 CHIP_SKU=00:00:00:D4 ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1 -c tools/kernel_flash/flash_l4t_nvme.xml -S 80GiB  -p "-c bootloader/t186ref/cfg/flash_t234_qspi.xml --no-systemimg" --no-flash  --massflash 5  --network usb0  jetson-orin-nano-devkit  external
```

### 1.3 Flash the JP5 to device
```
sudo tar xpf mfi_jetson-orin-nano-devkit.tar.gz
cd mfi_jetson-orin-nano-devkit
`sudo ./tools/kernel_flash/l4t_initrd_flash.sh --flash-only --massflash 1 --network usb0 --showlogs`
```

## 2 Generate the system image and OTA package of JP6.0
### 2.1 Prepare the SDK
```
cd jp6
wget https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/release/jetson_linux_r36.3.0_aarch64.tbz2
wget https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/release/tegra_linux_sample-root-filesystem_r36.3.0_aarch64.tbz2
wget https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/release/ota_tools_R36.3.0_aarch64.tbz2
```
### 2.2 Build system image of JP6
```
tar xf jetson_linux_r36.3.0_aarch64.tbz2
sudo tar xpf tegra_linux_sample-root-filesystem_r36.3.0_aarch64.tbz2 -C Linux_for_Tegra/rootfs
cd Linux_for_Tegra
sudo ./apply_binaries.sh
sudo BOARDID=3767 BOARDSKU=0001 FAB=300 BOARDREV=M.3 CHIP_SKU=00:00:00:D4 ./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1 -c tools/kernel_flash/flash_l4t_nvme.xml -S 80GiB -p "-c bootloader/generic/cfg/flash_t234_qspi.xml --no-systemimg" --no-flash --massflash 5 --network usb0 jetson-orin-nano-devkit external
```
### 2.3 Generate OTA payload package
```
cd jp6
tar xf ota_tools_R36.3.0_aarch64.tbz2
cd Linux_for_Tegra
sudo BASE_BSP=/home/<user>/jp5/Linux_for_Tegra ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh --external-device nvme0n1 -S 80GiB jetson-orin-nano-devkit R35-5
```
#### After the command finished. You will get the ota_payload_package.tar.gz in `Linux_for_Tegra/bootloader/jetson-orin-nano-devkit/ota_payload_package.tar.gz`
```
user@host:~/jp6/Linux_for_Tegra$ ls -lh bootloader/jetson-orin-nano-devkit/ota_payload_package.tar.gz
-rw-r--r-- 1 root root 2.7G Feb  7 02:48 bootloader/jetson-orin-nano-devkit/ota_payload_package.tar.gz
```
## 3 Apply OTA package on device
### 3.1 Prepare work
```
scp user@192.168.111.1:/home/user/jp6/Linux_for_Tegra/bootloader/jetson-orin-nano-devkit/ota_payload_package.tar.gz  ~
wget https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v3.0/release/ota_tools_R36.3.0_aarch64.tbz2
sudo mkdir /ota
sudo mv ~/ota_payload_package.tar.gz /ota
tar xf ota_tools_R36.3.0_aarch64.tbz2
```
### 3.2 Apply OTA
```
cd ~/Linux_for_Tegra/tools/ota_tools/version_upgrade
sudo ./nv_ota_start.sh /ota/ota_payload_package.tar.gz
```
#### After this command finished successfully. Reboot the device.

#### Finally, the device upgrade from 5.1.3 to 6.0 successfully.
Here is the system version info before and after OTA:
```
jp5@jp5-desktop:~$ cat /etc/nv_tegra_release
# R35 (release), REVISION: 5.0, GCID: 35550185, BOARD: t186ref, EABI: aarch64, DATE: Tue Feb 20 04:46:31 UTC 2024
```
```
jp6@jp6-desktop:~$ cat /etc/nv_tegra_release
# R36 (release), REVISION: 3.0, GCID: 36191598, BOARD: generic, EABI: aarch64, DATE: Mon May  6 17:34:21 UTC 2024
# KERNEL_VARIANT: oot
TARGET_USERSPACE_LIB_DIR=nvidia
TARGET_USERSPACE_LIB_DIR_PATH=usr/lib/aarch64-linux-gnu/nvidia
```

I attached the full uart log of my device while OTA from 5.1.3 to 6.0.

[5.1.3-to-6.0-ota-log.txt](https://github.com/user-attachments/files/18700631/5.1.3-to-6.0-ota-log.txt)

bigbearishappy, 2025-02-07:
I find a way to make it OTA from JP5.1.3 to JP6.1. 
Change `Linux_for_Tegra/tools/kernel_flash/flash_l4t_t234_nvme.xml` in JP6.1 's SDK:
```xml
-    <device type="external" instance="0" sector_size="512" num_sectors="EXT_NUM_SECTORS" >
+    <device type="external" instance="0" sector_size="512" num_sectors="200000000" >
```

Then generate the OTA package with following command:
```
cd jp6/Linux_for_Tegra/
sudo BASE_BSP=/path/to/jp5/Linux_for_Tegra/ ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh --external-device nvme0n1 -S 80GiB jetson-orin-nano-devkit R35-5
```

deadmerc, 2025-02-07:
Current status:

trying to make a package, my problem what JP5.1.3 devices was made with `p3509-a02+p3767-0000` config ( device already distributed, cant change that ), JP6.1 was made with `recomputer-orin-j401`, OTA tools dont really liked that, 
for JP5.13:
`ln -s p3509-a02+p3767-0000.conf jetson-orin-nano-devkit.conf`
for JP6.1:
`ln -s recomputer-orin-j401.conf jetson-orin-nano-devkit.conf`

will change ota scripts for pass this problem `The board name in OTA package(jetson-orin-nano-devkit) does not match current board(p3509-a02+p3767-0000)` and hopefully it will work

Thanks a lot for your assistance

deadmerc, 2025-02-11:
HI @henvas 

> we can run apt dist-upgrade safely on our device to upgrade from Jetpack 6.1 to 6.2?
You probably need to trick scripts, with `ln -s recomputer-orin-j401.conf jetson-orin-nano-devkit.conf`, then dist-upgrade can work, but it will use packages for `jetson-orin-nano-devkit` even dtb files are correct, so safely for sure not until Seeed confirm its compatible or APT repo provided, but it still can work partially or fully 🤷‍♂ 

> If not why is recomputer-orin-j401 used as board name when it just breaks OTA upgrading?
How I understood, thats separation are good, as custom carrier board have their own dtb files, problems become when OTA scripts ( in my case ) dont know anything about `recomputer-orin-j401` and stoped working, unfortunately the same with UEFI capsule update

deadmerc, 2025-02-11:
HI @bigbearishappy, 

So, my last attempt completely bricked device, also look like I have A/B enabled, is this correct?

```bash
$ ls -l /dev/disk/by-partlabel/
lrwxrwxrwx 1 root root 15 Feb 11 08:33 A_kernel -> ../../nvme0n1p2
lrwxrwxrwx 1 root root 15 Feb 11 08:33 A_kernel-dtb -> ../../nvme0n1p3
lrwxrwxrwx 1 root root 15 Feb 11 08:33 APP -> ../../nvme0n1p1
lrwxrwxrwx 1 root root 15 Feb 11 08:33 B_kernel -> ../../nvme0n1p5
lrwxrwxrwx 1 root root 15 Feb 11 08:33 B_kernel-dtb -> ../../nvme0n1p6
lrwxrwxrwx 1 root root 16 Feb 11 08:33 esp -> ../../nvme0n1p11
lrwxrwxrwx 1 root root 16 Feb 11 08:33 esp_alt -> ../../nvme0n1p14
lrwxrwxrwx 1 root root 15 Feb 11 08:33 recovery -> ../../nvme0n1p8
lrwxrwxrwx 1 root root 16 Feb 11 08:33 recovery_alt -> ../../nvme0n1p12
lrwxrwxrwx 1 root root 15 Feb 11 08:33 recovery-dtb -> ../../nvme0n1p9
lrwxrwxrwx 1 root root 16 Feb 11 08:33 recovery-dtb_alt -> ../../nvme0n1p13
lrwxrwxrwx 1 root root 16 Feb 11 08:33 RECROOTFS -> ../../nvme0n1p10
lrwxrwxrwx 1 root root 15 Feb 11 08:33 reserved_for_chain_A_user -> ../../nvme0n1p4
lrwxrwxrwx 1 root root 15 Feb 11 08:33 reserved_for_chain_B_user -> ../../nvme0n1p7
lrwxrwxrwx 1 root root 16 Feb 11 08:33 UDA -> ../../nvme0n1p15
```

If u have a chance, could try to flash first device with `p3509-a02+p3767-0000` J5.1.3 config ( like on Seeed guide page ), second device with `recomputer-orin-j401` J6.1 and make OTA update?

I believe UEFI capsule update are failing during OTA: 
 - after OTA UEFI remain the same
 - attempt to manually flash capsule UEFI updated failed, until Bootloader Update Payload ( BUP ) was generated with specific device spec file with `'boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=recomputer-orin-j401;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'`, only after that device got recognised

Currently I am trying to replace TEGRA_BL.Cap inside OTA package with disable sha hash verification, not sure if its right direction tho

bigbearishappy, 2025-02-12:
Sorry to hear this. I will flash device with `p3509-a02+p3767-0000` J5.1.3 config and then try to upgrade it to JP6.1 based on config `recomputer-orin-j401`

deadmerc, 2025-02-18:
Hi @bigbearishappy,

How are you doing? Maybe I can help with some problems which I already experienced during my tests?

Regards,
Max

bigbearishappy, 2025-02-18:
Still struggling with it. 
And I find the discussion which create by you in jetson's forum in [here](https://forums.developer.nvidia.com/t/image-based-ota-update-via-golden-image/320290/1).

I am trying to create a config file name with `p3509-a02+p3767-0000.conf` and then create a ota package based on it.

deadmerc, 2025-02-18:
> And I find the discussion which create by you in jetson's forum in [here](https://forums.developer.nvidia.com/t/image-based-ota-update-via-golden-image/320290/1).

Yeah, this case a bit special look like

> I am trying to create a config file name with p3509-a02+p3767-0000.conf and then create a ota package based on it.

I didnt fully understood, this config are in nvidia BSP, most probably was flashed via
```bash
./tools/kernel_flash/l4t_initrd_flash.sh --external-device nvme0n1p1 \
  -c tools/kernel_flash/flash_l4t_external.xml -p "-c bootloader/t186ref/cfg/flash_t234_qspi.xml" \
  --showlogs --network usb0 p3509-a02+p3767-0000 internal
```
from Seeed guide 5.1.2 ( and then `apt install nvidia-jetpack` will bring it to 5.1.3 state I believe )

https://wiki.seeedstudio.com/reComputer_J4012_Flash_Jetpack/#flash-to-jetson

bigbearishappy, 2025-02-18:
There is no `p3509-a02+p3767-0000.conf` in JP6.1 SDK. But in JP5.1.2 our device is flashed based on this config file. That's why I am trying to create `p3509-a02+p3767-0000.conf` in JP6.1 SDK.

deadmerc, 2025-02-18:
Oh, I understood now, ok

bigbearishappy, 2025-02-19:
@deadmerc 
I encountered with the problem same with you. I made a ota update package and apply it int target device.
But after reboot I get the following error log:
```
L4TLauncher: Attempting Direct Boot
EFI stub: Booting Linux Kernel...
EFI stub: ERROR: Invalid header detected on UEFI supplied FDT, ignoring ...
EFI stub: Generating empty DTB
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services...
```

I checked the uart log of the OTA process. And I found the OTA of rootfs is success(It show the version of 36.4.0 in `/etc/nv_tegra_release` in SSD after OTA finished.) but the UEFI update did't run. Here is a screenshot which shows the difference:

[image]

In the OTA process log. I find it enable the UEFI update by write some bytes in to a specific file. Here is the detail logs:
```
need_to_update_bootloader /mnt/ota_work
trigger_uefi_capsule_update /mnt/ota_work /dev/nvme0n1p
Mount esp partition on /opt/nvidia/esp
Copying /mnt/ota_work/TEGRA_BL.Cap into /opt/nvidia/esp/EFI/UpdateCapsule
Mount efivarfs on /sys/firmware/efi/efivars
Triggering UEFI capsule update by writing \x07\x00\x00\x00\x04\x00\x00\x00\x00\x00\x00\x00 to UEFI variable OsIndications-8be4df61-93ca-11d2-aa0d-00e098032b8c
dd if=/tmp/var_tmp.bin of=OsIndications-8be4df61-93ca-11d2-aa0d-00e098032b8c bs=12
1+0 records in
1+0 records out
12 bytes copied, 0.00576458 s, 2.1 kB/s
```
But in out condition, the UEFI update did't run. And you said you are trying to replace TEGRA_BL.Cap inside OTA package with disable sha hash verification. Any progress on you end?

deadmerc, 2025-02-19:
Hi @bigbearishappy 

> But in our condition, the UEFI update did't run. And you said you are trying to replace TEGRA_BL.Cap inside OTA package with disable sha hash verification. Any progress on you end?

I think I found the problem, but unable to resolve it

at first I tried to flash UEFI alone via capsule update, it just silently skip it ( there was no progress bar in serial log tho ), then I find out what capsule update I looking only for standard board names like `jetson-orin-nano-devkit`, so I guess when it see unknown board name like `p3509-a02+p3767-0000` it dont know what to do and just skip

for resolve it I create BUP specially targeted config file `tools/ota_tools/version_upgrade/ota_board_specs.conf` with new board name device `recomputer-orin-j401`, then it start working

```
jetson_orin_nano_devkit_ota_emmc_r35_r36_spec=(
        'boardid=3767;fab=000;boardsku=0003;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=jetson-orin-nano-devkit;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
        'boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=recomputer-orin-j401;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
        'boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=p3509-a02+p3767-0000;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
)
```

I tried to replicate the same logic when OTA package creating, in logs there also lines for which board names capsule update was made, with new lines inside `tools/ota_tools/version_upgrade/ota_board_specs.conf`, 2 new boards appeared ( I added both just in case )
 - p3509-a02+p3767-0000
 - recomputer-orin-j401
So thats should work, because correct TEGRA_BL.Cap should be inside package without tampering,  but problem remain the same and I am not sure why, please try to add new specs inside your spec file then check if capsule update was made for `p3509-a02+p3767-0000`, maybe I missed something

deadmerc, 2025-02-19:
Maybe worst to check UEFI update alone at `p3509-a02+p3767-0000` J513, I was able to pull it off only with `recomputer-orin-j401` J61, maybe there is something special required

deadmerc, 2025-02-19:
> In the OTA process log. I find it enable the UEFI update by write some bytes in to a specific file. Here is the detail logs:

How I understand from NVIDIA guide, thats just signal to system for attempt UEFI update of reboot

bigbearishappy, 2025-02-20:
As we still work on this problem and no significant progress. 
I want to know if you considered preparing a backup plan. 
You can update the apt source to R36.4.0 and then do a `apt dis-upgrade`.
I already tried it. It seems that the system app are updated to the latest version. But the kernel version is still same with JP5.1.3.

bigbearishappy, 2025-02-20:
I flash the UEFI firmware of JP5.1.3 to my device with the debug log opened.
Then I reproduce this problem and I got the following log while running UEFI:
```
secondary_gpt_backup not found in package: Not Found
FmpDxe(NVIDIA System Firmware): CheckTheImage() - FmpDeviceLib CheckImage failed. Status = Aborted
FmpDxe(NVIDIA System Firmware): SetTheImage() - Check The Image failed with Aborted
```

Have you took look into this?

deadmerc, 2025-02-20:
> Have you took look into this?

First time see such errors, could you please send me `p3509-a02+p3767-0000` config, I will try to create new OTA package and debug UEFI problem

It was capsule update UEFI from 5.1.3 to 6.1 without updating rootfs?

> I want to know if you considered preparing a backup plan.
You can update the apt source to R36.4.0 and then do a apt dis-upgrade.

Unfortunately no, I proposed such solution to upper management long time ago but it was rejected

deadmerc, 2025-02-20:
During OTA package creation, did you saw creation BUP for `p3509-a02+p3767-0000`?

This specific part
```
Generate Multi-Spec BUP Tool

boardid=3767;fab=000;boardsku=0003;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=jetson-orin-nano-devkit;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT ,  == jetson-orin-nano-devkit
Info. BUP entry check is done.
```

Full
```
SUCCESS: generate binaries for OTA "boardid=3767;fab=000;boardsku=0003;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=jetson-orin-nano-devkit;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT"
SUCCESS: generate binaries for OTA "boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=recomputer-orin-j401;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT"
SUCCESS: generate binaries for OTA "boardid=3767;fab=300;boardsku=0003;boardrev=N.2;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=p3509-a02+p3767-0000;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT"

./external_device/
./external_device/images-R36-ToT/
./external_device/images-R36-ToT/3767--0003-/
./external_device/images-R36-ToT/3767--0003-/recovery.img
./external_device/images-R36-ToT/3767--0003-/tegra234-j401-p3768-0000+p3767-0003-recomputer.dtb.rec
./external_device/images-R36-ToT/3767--0003-/flash.idx
./external_device/images-R36-ToT/3767--0003-/kernel_tegra234-j401-p3768-0000+p3767-0003-recomputer.dtb
./external_device/images-R36-ToT/3767--0003-/boot.img
./external_device/images-R36-ToT/3767-300-0003-N.2/
./external_device/images-R36-ToT/3767-300-0003-N.2/kernel_tegra234-p3768-0000+p3767-0003-nv.dtb
./external_device/images-R36-ToT/3767-300-0003-N.2/tegra234-p3768-0000+p3767-0003-nv.dtb.rec
./external_device/images-R36-ToT/3767-300-0003-N.2/recovery.img
./external_device/images-R36-ToT/3767-300-0003-N.2/tegra234-j401-p3768-0000+p3767-0003-recomputer.dtb.rec
./external_device/images-R36-ToT/3767-300-0003-N.2/flash.idx
./external_device/images-R36-ToT/3767-300-0003-N.2/kernel_tegra234-j401-p3768-0000+p3767-0003-recomputer.dtb
./external_device/images-R36-ToT/3767-300-0003-N.2/boot.img
./external_device/system.img
./external_device/system.img.sha1sum
./ota_backup_files_list.txt
Copy NVIDIA provided rootfs updater to /media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp/nv_ota_rootfs_updater.sh
/media/saschao/Storage/seeed/Linux_for_Tegra/tools/ota_tools/version_upgrade /media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp /media/saschao/Storage/seeed/Linux_for_Tegra
'nv_ota_validate.sh' -> '/media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp/nv_ota_validate.sh'
'nv_ota_update.sh' -> '/media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp/nv_ota_update.sh'
'nv_ota_common_utils.func' -> '/media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp/nv_ota_common_utils.func'
'nv_ota_update_rootfs_in_recovery.sh' -> '/media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp/nv_ota_update_rootfs_in_recovery.sh'
/media/saschao/Storage/seeed/Linux_for_Tegra/ota_base_dir_tmp /media/saschao/Storage/seeed/Linux_for_Tegra
Generate BUP file by running command: ROOTFS_AB= ROOTFS_ENC=  /media/saschao/Storage/seeed/Linux_for_Tegra/l4t_generate_soc_bup.sh -f /tmp/board_spec_file -e tmp_board_spec -b jetson-orin-nano-devkit t23x

Generate Multi-Spec BUP Tool

boardid=3767;fab=000;boardsku=0003;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=jetson-orin-nano-devkit;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT ,  == jetson-orin-nano-devkit
Info. BUP entry check is done.
###############################################################################
# L4T BSP Information:
# R36 , REVISION: 4.0
# User release: 0.0
###############################################################################
```

bigbearishappy, 2025-02-20:
> > Have you took look into this?
> 
> First time see such errors, could you please send me `p3509-a02+p3767-0000` config, I will try to create new OTA package and debug UEFI problem

I didn't create `p3509-a02+p3767-0000` for JP6.1, my debug work is based on the `jetson-orin-nano-devkit`.

> 
> It was capsule update UEFI from 5.1.3 to 6.1 without updating rootfs?

No I update the UEFI with rootfs updating too.

> 
> > I want to know if you considered preparing a backup plan.
> > You can update the apt source to R36.4.0 and then do a apt dis-upgrade.
> 
> Unfortunately no, I proposed such solution to upper management long time ago but it was rejected

bigbearishappy, 2025-02-20:
> During OTA package creation, did you saw creation BUP for `p3509-a02+p3767-0000`?
>
> This specific part
>
> ```
> Generate Multi-Spec BUP Tool [...]

Actually, I am now working on find out the reason why current UEFI firmware just skip the update request.

deadmerc, 2025-02-20:
> Actually, I am now working on find out the reason why current UEFI firmware just skip the update request.

How I understood, the whole problem in board name `jetson-orin-nano-devkit` dont match with `p3509-a02+p3767-0000`, I am guessing its crashing somewhere there `/opt/nvidia/l4t-bootloader-config/nv-l4t-bootloader-config.sh`

deadmerc, 2025-03-04:
Hi @bigbearishappy, do you have any news?

bigbearishappy, 2025-03-04:
> > Then I tried to add seeed's board recomputer-orin to the OTA packge generation scripts, and I can make it work.
> 
> Sorry, English is not my native language, to understand, you are currently trying to change OTA scripts to make `p3509-a02+p3767-0000` compatible with new config `recomputer-orin`?

I work with `recomputer-orin` just for verifying the customization of OTA package generation scripts. It has nothing to do with `p3509-a02+p3767-0000.conf`.
I create `p3509-a02+p3767-0000.conf` with command `ln -s recomputer-orin-j401.conf  p3509-a02+p3767-0000.conf`.

deadmerc, 2025-03-04:
Understood, thank you for update

baorepo, 2025-03-13:
@deadmerc  I am very sorry to keep you waiting. I have verified the upgrade process of ***p3509-a02-p3767-0000*** from 5.1.3 -> 6.1, and some modifications need to be made to Jetpack and ota_tools. Later @bigbearishappy  will also integrate these changes into the source code.  Specific OTA process, this step is still valid.  https://github.com/Seeed-Studio/Linux_for_Tegra/issues/23#issuecomment-2641840323  If you have any questions, please feel free to ask, Bear and I will do our best to serve you.
```patch
--- nvmassfusegen.sh	2024-09-13 12:29:35.802862036 +0800
+++ ../../Linux_for_Tegra/nvmassfusegen.sh	2025-03-12 16:22:27.013499206 +0800
@@ -1211,6 +1211,7 @@
 	"jetson-agx-xavier-devkit" \
 	"jetson-agx-xavier-industrial" \
 	"jetson-orin-nano-devkit" \
+	"p3509-a02+p3767-0000" \
 	"jetson-xavier-nx-devkit-emmc" \
 );
 
--- nvautoflash.sh	2024-09-13 12:29:35.818861906 +0800
+++ ../../Linux_for_Tegra/nvautoflash.sh	2025-03-13 11:20:21.593379750 +0800
@@ -486,6 +486,10 @@
 	"jetson-orin-nano-devkit"		"internal"	"3767"	"0003" \
 	"jetson-orin-nano-devkit"		"internal"	"3767"	"0004" \
 	"jetson-orin-nano-devkit"		"internal"	"3767"	"0005" \
+	"p3509-a02+p3767-0000"		"internal"	"3767"	"0000" \
+	"p3509-a02+p3767-0000"		"internal"	"3767"	"0001" \
+	"p3509-a02+p3767-0000"		"internal"	"3767"	"0003" \
+	"p3509-a02+p3767-0000"		"internal"	"3767"	"0004" \
 	"jetson-agx-orin-devkit-industrial" "internal" 	"3701"	"0008"
 
 
--- jetson_board_spec.cfg	2024-09-13 12:29:35.846861682 +0800
+++ ../../Linux_for_Tegra/jetson_board_spec.cfg	2025-03-13 11:24:22.990909463 +0800
@@ -67,6 +67,17 @@
     'boardid=3767;fab=000;boardsku=0005;boardrev=;chiprev=;chipsku=00:00:00:D5;board=jetson-orin-nano-devkit;rootdev=mmcblk0p1;bup_type=bl'
     # orin-nano 4GB
     'boardid=3767;fab=000;boardsku=0004;boardrev=;chiprev=;chipsku=00:00:00:D6;board=jetson-orin-nano-devkit;rootdev=mmcblk0p1;bup_type=bl'
+
+    # orin-nx 16GB
+    'boardid=3767;fab=000;boardsku=0000;boardrev=;chiprev=;chipsku=00:00:00:D5;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=bl'
+    # orin-nx 8GB
+    'boardid=3767;fab=000;boardsku=0001;boardrev=;chiprev=;chipsku=00:00:00:D6;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=bl'
+
+    # orin-nano 8GB
+    'boardid=3767;fab=000;boardsku=0003;boardrev=;chiprev=;chipsku=00:00:00:D5;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=bl'
+    'boardid=3767;fab=000;boardsku=0005;boardrev=;chiprev=;chipsku=00:00:00:D5;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=bl'
+    # orin-nano 4GB
+    'boardid=3767;fab=000;boardsku=0004;boardrev=;chiprev=;chipsku=00:00:00:D6;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=bl'
 )
 
 t23x_bl_spec=(
@@ -109,6 +120,19 @@
     'boardid=3767;fab=000;boardsku=0005;boardrev=;chiprev=;board=jetson-orin-nano-devkit;rootdev=mmcblk0p1;bup_type=kernel'
     # orin-nano 4GB
     'boardid=3767;fab=000;boardsku=0004;boardrev=;chiprev=;board=jetson-orin-nano-devkit;rootdev=mmcblk0p1;bup_type=kernel'
+
+
+    # orin-nx 16GB
+    'boardid=3767;fab=000;boardsku=0000;boardrev=;chiprev=;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=kernel'
+    # orin-nx 8GB
+    'boardid=3767;fab=000;boardsku=0001;boardrev=;chiprev=;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=kernel'
+
+    # orin-nano 8GB
+    'boardid=3767;fab=000;boardsku=0003;boardrev=;chiprev=;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=kernel'
+    'boardid=3767;fab=000;boardsku=0005;boardrev=;chiprev=;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=kernel'
+    # orin-nano 4GB
+    'boardid=3767;fab=000;boardsku=0004;boardrev=;chiprev=;board=p3509-a02+p3767-0000;rootdev=mmcblk0p1;bup_type=kernel'
+
 )
 
 # FIXME: k5.15 uses non-public DT files for holoscan hence BUP test cannot support the platforms
--- tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh	2024-09-13 12:45:47.299045237 +0800
+++ ../../Linux_for_Tegra/tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh	2025-03-13 11:25:26.589552271 +0800
@@ -78,6 +78,7 @@
 	'jetson-agx-orin-devkit:nvme0n1'
 	'jetson-agx-orin-devkit-industrial:nvme0n1'
[... 72 lines cut ...]
+
+    # orin-nano 8GB
+    'boardid=3767;fab=000;boardsku=0003;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=p3509-a02+p3767-0000;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
+	'boardid=3767;fab=000;boardsku=0005;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D5;board=p3509-a02+p3767-0000;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
+    # orin-nano 4GB
+    'boardid=3767;fab=000;boardsku=0004;boardrev=;fuselevel_s=1;chiprev=;chipsku=00:00:00:D6;board=p3509-a02+p3767-0000;rootdev=nvme0n1p1;bup_type=bl;signed_img_dir=images-R36-ToT'
+)
+
+P3509_A02_P3767_0000_R35_5_ALIAS="p3509_a02_p3767_0000_ota_emmc_r35_r36_spec"
+P3509_A02_P3767_0000_R36_4_ALIAS="p3509_a02_p3767_0000_ota_emmc_r35_r36_spec"
+
+
 # List the supported t23x devices
 T23X_DEVICES=(
-	'JETSON_AGX_ORIN_DEVKIT'
-	'JETSON_AGX_ORIN_DEVKIT_INDUSTRIAL'
-	'JETSON_ORIN_NANO_DEVKIT'
-)
+    'JETSON_AGX_ORIN_DEVKIT'
+    'JETSON_AGX_ORIN_DEVKIT_INDUSTRIAL'
+    'JETSON_ORIN_NANO_DEVKIT'
+    'P3509_A02_P3767_0000'
+)

```

deadmerc, 2025-03-14:
Hi @baorepo, thank you very much for update, I will try to test it on our device as soon as possible

deadmerc, 2025-03-19:
Hi guys, I have some difficulties, 

My command is 
```
BOARDID=3767 BOARDSKU=0003 FAB=300 BOARDREV=N.2 CHIP_SKU=00:00:00:D5 ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh -s -f /media/saschao/Storage/system.img.raw --external-device nvme0n1 -S 118GiB p3509-a02+p3767-0000 R35-5
```
Config `p3509-a02+p3767-0000` symlink to `recomputer-orin-j401` is that correct?

OTA package generated but doesnt contain rootfs image ( just 39mb )
```
# ls -lah bootloader/p3509-a02+p3767-0000/ota_payload_package.tar.gz
-rw-r--r-- 1 root root 39M Mar 18 06:42 bootloader/p3509-a02+p3767-0000/ota_payload_package.tar.gz
```

My rootfs image was copied from another already installed device ( instead auto generated empty image without -s, as there is no our libs installed )

```
ls -lah /media/saschao/Storage/system.img.raw
-rwxrwxrwx 1 root root 118G Mar 14 08:34 /media/saschao/Storage/system.img.raw
```

Full log attached

[generate_ota_package_2025-03-18_06-26-40.txt](https://github.com/user-attachments/files/19336497/generate_ota_package_2025-03-18_06-26-40.txt)

Is there something else required to support golden-image method?

baorepo, 2025-03-19:
@deadmerc  This is the step to generate OTA files, the way you generate, I have not tried. 
[image]
refer to https://github.com/Seeed-Studio/Linux_for_Tegra/issues/23#issuecomment-2641840323

baorepo, 2025-03-19:
and p3509-a02+p3767-0000.conf -> p3509-a02-p3767-0000.conf.

[image]

deadmerc, 2025-03-19:
Understood, thank you, I will try to make it run with auto-generated image first and fixed config file

baorepo, 2025-03-25:
@deadmerc  Any updates? Did the OTA work? Is this issue close to being closed?

deadmerc, 2025-03-25:
Hi @bigbearishappy, sorry for the delay, another emergency appeared in front, I will try to update you today. If you have time, could you please try to use golden image instead of auto-generated? thats guide from nvidia, is our end-goal

```
tools/ota_tools/version_upgrade/Image_based_OTA_Examples.txt

4. Using golden image in OTA update.
Customers might flash a device with the latest released BSP and then make a golden image
dependent on this device. They can deploy the golden image on other devices that were
flashed with the previously released BSP through an OTA update.

The "l4t_generate_ota_package.sh" script includes the "-f" and "-o" options so that
customers can use their own rootfs updater and update the rootfs partition with their
own rootfs image (for example, a golden image).

However, for some customers, the golden image is a raw partition image that is
directly obtained from rootfs partition. (These customers might not
want to write their own rootfs updater.)

The following examples provide the steps for using this golden image in an image-based OTA.

Case 12: rootfs A/B disabled, disk encryption disabled:
  a. Put the target device into recovery mode.
  b. Boot the device to initrd by running the following command:
     $ sudo ./tools/kernel_flash/l4t_initrd_flash.sh --initrd <target board> mmcblk0p1
  c. Enter bash shell on the target device.
  d. Mount a USB drive or network storage on the target device.
  e. Generate golden image in the mounted storage device by running the
     following command:
     $ sudo dd if=<rootfs partition> of=<mount ponint>/golden.img.raw
     The "<rootfs partition>" is /dev/mmcblk0p1 for the internal eMMC storage device
     or /dev/nvme0n1p1 for the external NVMe storage device.
  f. Copy the golden image into the TARGET_BSP directory on the host machine by running
     the following command:
     $ sudo cp golden.img.raw ${TARGET_BSP}/bootloader/system.img.raw
  g. Add the "-s" option when generating the OTA payload package.
     For the internal eMMC storage device, running the following command:
     $ sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh \
       -s <target board> <base version>
     For the external NVMe storage device, running the following command:
     $ sudo ./tools/ota_tools/version_upgrade/l4t_generate_ota_package.sh \
       -s --external-device nvme0n1 -S <rootfs size > <target board> <base version>

```

deadmerc, 2025-03-25:
@baorepo update, OTA with generated image are working, remain last step, using our image with application and compiled libs instead of empty one like in case 12 from documentation

[image]

baorepo, 2025-04-01:
@deadmerc  Great. If you have any questions, feel free to keep talking to us.

deadmerc, 2025-04-01:
> Oh, I just noticed that it seems to be a different OTA. Why do you have to use this one?

After basic OTA we have like fresh installation, without anything, means we have to install and compile all our stuff into each device again, thats fine if we have just couple devices. But when u have 2k+ distributed to clients with weak network connection it is nightmare. We can also disturb internal work.

So solution is, create one J6.1 device, install everything and compile all required libraries. Extract rootfs, create OTA file which will contain our application, upload inside client internal network, then trigger OTA overnight and job is done. Thats why we need OTA with custom image instead of empty one

baorepo, 2025-04-01:
@deadmerc  In fact, you can also install your application here by modifying rootfs via qemu. And generated OTA packages, but also incremental system updates. OTA packages will be smaller.
[image]
Maybe I still don't understand your needs, I understand https://github.com/Seeed-Studio/Linux_for_Tegra/issues/23#issuecomment-2641840323  is incremental update,  https://github.com/Seeed-Studio/Linux_for_Tegra/issues/23#issuecomment-2750226925 is full update.

deadmerc, 2025-04-01:
I am not sure if incremental update will work, as a lot of changed from system libraries point of view + some custom stuff, in theory its possible to manually modify rootfs to copy everything, but I cant really imagine how to do it without forgetting some small pieces, as its scattered all over the system. 

Thats why I just to want to take main partition and essentially replace old one with new ( full update = OTA via golden image? )

baorepo, 2025-04-01:
@deadmerc  All right. You see, I understand, right? This is developed and deployed on a device, but it is more difficult to reproduce the device. You need to copy the original device in bulk and update all the devices.

deadmerc, 2025-04-01:
> You need to copy the original device in bulk and update all the devices.

Yeah, thats what I am trying to achieve via OTA update, with latest changes it do not allow we to use my own rootfs image extracted from already installed device for some reason, `-s` argument

[17 more replies]

closed 2025-04-11 by deadmerc