Seeed-Studio/Linux_for_Tegra #26: Jetpack 6.1 Not Exposing SSH Over USB?

dduuukk, 2025-02-03:
I have followed the steps to flash r36.4.0 with the J4011, but noticed that once flashed, the device is not successfully exposing an SSH server over the USB-C port. I can see in dmesg that it tries and then shuts down. However, it appears that the compiled system image from the [wiki](https://wiki.seeedstudio.com/reComputer_J4012_Flash_Jetpack/) does expose SSH over USB-C. 

I can see when flashing that initrd finds the board SKU to be 1. I can also see that for a board SKU of 1, the only changes made to the default device tree are in `tegra234-j401-p3768-0000+p3767-recomputer-common.dts`, where there is the added micro-usb stuff. I guess I am confused because the J4011 does not have a micro-usb.

My end goal is to create a custom rootfs I can flash to the devices, so I would prefer to use source code rather than a compiled system image. Am I missing something that would allow the device to expose an SSH port over USB once flashed?

bigbearishappy, 2025-02-05:
The quickest way to fix this problem is to run `depmod -a` on the target device. 
I will upload the fix code to branch R36.4.0 ASAP.

bigbearishappy, 2025-02-05:
I checked the code of branch r36.4.0. And the code is already [there](https://github.com/Seeed-Studio/Linux_for_Tegra/blob/9dac930174842938be7e90233bc66ff14cdd2a33/.gitlab-ci.yml#L124).
You can have a try and compile the system image again.

dduuukk, 2025-02-05:
Recompiling with depmod fixed my issue. Thanks!

closed 2025-02-05 by dduuukk