Seeed-Studio/Linux_for_Tegra #28: Is there any support for the Orin AGX H01 carrier board?

imed-twosix, 2025-02-20:
I am trying to build a yocto image for the Orin AGX 32GB H01 Kit. I can get the device tree files and additional systemd services from the H01 Wiki, but not sure how to compile the kernel image without BSP.

Thanks

imed-twosix, 2025-02-21:
Hi @bigbearishappy. I am able to build a yocto image for the AGX Orin Nvidia Devkit just fine, but I want to build it for the AGX Orin H01 Kit. 
The [H01 wiki instructions for a Jetpack build](https://wiki.seeedstudio.com/Jetson_AGX_Orin_32GB_H01_Flash_Jetpack/#tech-support--product-discussion) overwrite the Nvidia files with the below Seeed H01 BSP files: 

```
imed@ubuntu:~/Documents/jetson$ tree 605_jetpack6.1
605_jetpack6.1
└── Linux_for_Tegra
    ├── bootloader
    │   ├── generic
    │   │   └── BCT
    │   │       ├── tegra234-mb1-bct-pinmux-p3701-0000-a04.dtsi
    │   │       └── tegra234-mb1-bct-pinmux-p3767-dp-a03.dtsi
    │   ├── tegra234-mb1-bct-gpio-p3701-0000-a04.dtsi
    │   ├── tegra234-mb1-bct-gpio-p3767-dp-a03.dtsi
    │   └── tegra234-mb2-bct-common.dtsi
    ├── kernel
    │   ├── dtb
    │   │   ├── tegra234-p3737-0000+p3701-0000-dynamic.dtbo
    │   │   ├── tegra234-p3737-0000+p3701-0000-nv.dtb
    │   │   ├── tegra234-p3737-0000+p3701-0004-nv.dtb
    │   │   ├── tegra234-p3737-0000+p3701-0005-nv.dtb
    │   │   └── tegra234-p3737-0000+p3701-0008-nv.dtb
    │   └── Image
    ├── p3737-0000-p3701-0000.conf
    └── rootfs
        ├── etc
        │   └── systemd
        │       └── system
        │           ├── multi-user.target.wants
        │           │   └── startup.service -> ../startup.service
        │           └── startup.service
        ├── leetop
        │   ├── code_spkmic.sh
        │   ├── quectel
        │   │   ├── quectel-CM
        │   │   └── quectel.sh
        │   └── startup.sh
        └── usr
            └── lib
                └── modules
                    └── 5.15.148-tegra
                        ├── kernel
                        │   └── drivers
                        │       ├── net
                        │       │   ├── usb
                        │       │   │   ├── cdc_mbim.ko
                        │       │   │   └── cdc_ncm.ko
                        │       │   └── wireless
                        │       │       └── intel
                        │       │           └── iwlwifi
                        │       │               ├── dvm
                        │       │               │   └── iwldvm.ko
                        │       │               ├── iwlwifi.ko
                        │       │               └── mvm
                        │       │                   └── iwlmvm.ko
                        │       └── usb
                        │           ├── class
                        │           │   └── cdc-wdm.ko
                        │           └── serial
                        │               ├── option.ko
                        │               └── usb_wwan.ko
                        └── updates
                            └── drivers
                                └── media
                                    ├── i2c
                                    │   ├── imx390yuv.ko
                                    │   ├── max9295_yuv.ko
                                    │   └── max9296_yuv.ko
                                    └── platform
                                        └── tegra
                                            └── camera
                                                └── tegra-camera.ko

36 directories, 30 files
```
I can patch my Yocto build with most of these files, but I am not sure how or if I should replace the compiled kernel Image binary in `605_jetpack6.1/Linux_for_Tegra/kernel/Image`
What am I missing here?

baorepo, 2025-02-22:
@imed-twosix  meta-tegra has nothing to do with this repository, if you need to flash it, please refer to this document. https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit

bigbearishappy, 2025-02-24:
> Hi [@bigbearishappy](https://github.com/bigbearishappy). I am able to build a yocto image for the AGX Orin Nvidia Devkit just fine, but I want to build it for the AGX Orin H01 Kit. The [H01 wiki instructions for flashing Jetpack](https://wiki.seeedstudio.com/Jetson_AGX_Orin_32GB_H01_Flash_Jetpack/#tech-support--product-discussion) show the difference between Nvidia Devkit and Seed H01 Kit:
> 
> ```
> imed@ubuntu:~/Documents/jetson$ tree 605_jetpack6.1
> 605_jetpack6.1
> └── Linux_for_Tegra
>     ├── bootloader
>     │   ├── generic
>     │   │   └── BCT
>     │   │       ├── tegra234-mb1-bct-pinmux-p3701-0000-a04.dtsi
>     │   │       └── tegra234-mb1-bct-pinmux-p3767-dp-a03.dtsi
>     │   ├── tegra234-mb1-bct-gpio-p3701-0000-a04.dtsi
>     │   ├── tegra234-mb1-bct-gpio-p3767-dp-a03.dtsi
>     │   └── tegra234-mb2-bct-common.dtsi
>     ├── kernel
>     │   ├── dtb
>     │   │   ├── tegra234-p3737-0000+p3701-0000-dynamic.dtbo
>     │   │   ├── tegra234-p3737-0000+p3701-0000-nv.dtb
>     │   │   ├── tegra234-p3737-0000+p3701-0004-nv.dtb
>     │   │   ├── tegra234-p3737-0000+p3701-0005-nv.dtb
>     │   │   └── tegra234-p3737-0000+p3701-0008-nv.dtb
>     │   └── Image
>     ├── p3737-0000-p3701-0000.conf
>     └── rootfs
>         ├── etc
>         │   └── systemd
>         │       └── system
>         │           ├── multi-user.target.wants
>         │           │   └── startup.service -> ../startup.service
>         │           └── startup.service
>         ├── leetop
>         │   ├── code_spkmic.sh
>         │   ├── quectel
>         │   │   ├── quectel-CM
>         │   │   └── quectel.sh
>         │   └── startup.sh
>         └── usr
>             └── lib
>                 └── modules
>                     └── 5.15.148-tegra
>                         ├── kernel
>                         │   └── drivers
>                         │       ├── net
>                         │       │   ├── usb
>                         │       │   │   ├── cdc_mbim.ko
>                         │       │   │   └── cdc_ncm.ko
>                         │       │   └── wireless
>                         │       │       └── intel
>                         │       │           └── iwlwifi
>                         │       │               ├── dvm
>                         │       │               │   └── iwldvm.ko
>                         │       │               ├── iwlwifi.ko
>                         │       │               └── mvm
>                         │       │                   └── iwlmvm.ko
>                         │       └── usb
>                         │           ├── class
>                         │           │   └── cdc-wdm.ko
>                         │           └── serial
>                         │               ├── option.ko
>                         │               └── usb_wwan.ko
>                         └── updates
>                             └── drivers
>                                 └── media
>                                     ├── i2c
>                                     │   ├── imx390yuv.ko
>                                     │   ├── max9295_yuv.ko
>                                     │   └── max9296_yuv.ko
>                                     └── platform
>                                         └── tegra
>                                             └── camera
>                                                 └── tegra-camera.ko
> 
> 36 directories, 30 files
> ```
> 
> I can patch my Yocto build with most of these files, but I am not sure how or if I should replace the compiled kernel Image binary in `605_jetpack6.1/Linux_for_Tegra/kernel/Image` What am I missing here?

You should find out the difference between official Image and the Image from `605_jetpack6.1`, and then make the change in yocto's kernel config file.

imed-twosix, 2025-02-26:
I created a tegra-binaries_36.4.0.bbappend file that copies the 605_jetpack6.1 files into Yocto, and now my board boots fine. I can verify it is producing the correct bootloader /dtsi files but the Seeed Studio kernel/dtbo files are not being used. And when after I boot the USB and ETH1 are not working

bigbearishappy, 2025-02-27:
Let's take the orin-nx for example. As we can see from the [config](https://github.com/OE4T/meta-tegra/blob/2bbc15ae8b5aa1dc570eae544cd15d83ce239a9a/conf/machine/include/orin-nx.inc#L20C23-L20C60) of orin nx. It speicfied the main dtb name of the board.
So you can create a new board for you hardware, and replace the main dtb with the new one.

imed-twosix, 2025-02-27:
I figured it out. For anyone else who is trying to add the custom BSP to yocto, you will want to copy the `bootloader` and `kernel` folders from the `A605_jetpack6.1.tar` to your custom yocto layer. Here is how I did it:
```
pokyuser@36916de4cb79:~$ tree meta-mylayer
meta-mylayer
└── recipes-bsp
    └── seeed-studio-h01
        ├── files
        │   ├── bootloader
        │   │   ├── generic
        │   │   │   └── BCT
        │   │   │       ├── tegra234-mb1-bct-pinmux-p3701-0000-a04.dtsi
        │   │   │       └── tegra234-mb1-bct-pinmux-p3767-dp-a03.dtsi
        │   │   ├── tegra234-mb1-bct-gpio-p3701-0000-a04.dtsi
        │   │   ├── tegra234-mb1-bct-gpio-p3767-dp-a03.dtsi
        │   │   └── tegra234-mb2-bct-common.dtsi
        │   └── kernel
        │       ├── tegra234-p3737-0000+p3701-0000-dynamic.dtbo
        │       ├── tegra234-p3737-0000+p3701-0000-nv.dtb
        │       ├── tegra234-p3737-0000+p3701-0004-nv.dtb
        │       ├── tegra234-p3737-0000+p3701-0005-nv.dtb
        │       └── tegra234-p3737-0000+p3701-0008-nv.dtb
        ├── nvidia-kernel-oot-dtb_36.4.0.bbappend
        └── tegra-binaries_36.4.0.bbappend
7 directories, 12 files

pokyuser@36916de4cb79:~$ cat meta-mylayer/recipes-bsp/seeed-studio-h01/nvidia-kernel-oot-dtb_36.4.0.bbappend 
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SRC_URI:append = " file://kernel/"
do_install:append() {
    cp -f ${WORKDIR}/kernel/* ${STAGING_DIR_HOST}/boot/devicetree/
}

pokyuser@36916de4cb79:~$ cat meta-mylayer/recipes-bsp/seeed-studio-h01/tegra-binaries_36.4.0.bbappend 
FILESEXTRAPATHS:prepend := "${THISDIR}/files:"
SRC_URI:append = " file://bootloader/"
do_preconfigure:append() {
    cp -rf ${WORKDIR}/bootloader/* ${S}/bootloader/
}
```