Seeed-Studio/Linux_for_Tegra #31: Jetpack 6.2 (r36.4.3)

will4ntz, 2025-02-25:
When will the source for jetpack 6.2 (r36.4.3) be available? I need it quickly to customize my rootfs

bigbearishappy, 2025-02-26:
The jetpack 6.2 software for seeed is still under development. We will release it after the test work is finished.
You can have a try to port the code of jetpack 6.1 to 6.2. Most of them are the same. 
And don't forget to merge the super mode board config file to recomputer.

will4ntz, 2025-02-27:
Ok, I'm using JP6.2 (r36.4.3) coming from the nvidia repository, carrying the files from this repository, with DTB_FILE=tegra234-p3768-0000+p3767-0003-nv-super.dtb, but using jtop the j3011 computer does not activate the clocks according to the "super" configuration cpu 1.7Ghz and gpu 1,020MHz, is a custom dtb necessary for the j301x computer?

[image]

bigbearishappy, 2025-02-28:
> Ok, I'm using JP6.2 (r36.4.3) coming from the nvidia repository, carrying the files from this repository, with DTB_FILE=tegra234-p3768-0000+p3767-0003-nv-super.dtb, but using jtop the j3011 computer [...]

No, the custom dts for j301x recomputer is specially for the peripherals of j301x.
According to you screenshot you provided. It seems that you don't enable the MAXN mode of you device. And the system power is just 7.2W.
If you want to test the MAXN mode of your device. You can have a try with the help from [here](https://elinux.org/Jetson/L4T/TRT_Customized_Example#Maximize_the_Device_Performance)

will4ntz, 2025-02-28:
OK, even using jp6.2 from sdkmanager, activating jetson_clocks, and nvpmode 0, I got the same result :(

will4ntz, 2025-02-28:
Update: 

After using the recomputer-orin-j401.conf file as a reference, recomputer-orin-j401-super.conf was created with the content below:

```

source "${LDK_DIR}/p3768-0000-p3767-0000-a0.conf";

update_flash_args_p3768_0000_p3767_0000()
{
	if [ "${board_sku}" = "0004" ]; then
                BPFDTB_FILE="tegra234-bpmp-3767-0004-3768-super.dtb";
		DTB_FILE=tegra234-p3768-0000+p3767-0004-nv-super.dtb;
	elif [ "${board_sku}" = "0003" ]; then
		BPFDTB_FILE="tegra234-bpmp-3767-0003-3768-super.dtb";
		DTB_FILE=tegra234-p3768-0000+p3767-0003-nv-super.dtb;
	elif [ "${board_sku}" = "0000" ]; then
                BPFDTB_FILE="tegra234-bpmp-3767-0000-3768-super.dtb";
		DTB_FILE=tegra234-p3768-0000+p3767-0000-nv-super.dtb;
	elif [ "${board_sku}" = "0001" ]; then
                BPFDTB_FILE="tegra234-bpmp-3767-0001-3768-super.dtb";
		DTB_FILE=tegra234-p3768-0000+p3767-0001-nv-super.dtb;
	else
		echo "Error: Unrecognized module SKU ${board_sku}";
		exit 1;
	fi

	TBCDTB_FILE="${DTB_FILE}";
}

update_flash_args()
{
	update_flash_args_common
	update_flash_args_p3768_0000_p3767_0000
}

PINMUX_CONFIG="tegra234-mb1-bct-pinmux-p3767-hdmi-a03.dtsi";
PMC_CONFIG="tegra234-mb1-bct-padvoltage-p3767-hdmi-a03.dtsi";
OVERLAY_DTB_FILE+=",tegra234-dcb-p3767-0000-hdmi.dtbo,tegra234-p3767-camera-p3768-imx219-dual-seeed.dtbo";
DCE_OVERLAY_DTB_FILE="tegra234-dcb-p3767-0000-hdmi.dtbo";
```

After the changes it worked perfectly

Thanks @bigbearishappy for helping

closed 2025-02-28 by will4ntz

fettahyildizz, 2025-05-23:
Hello, may I ask why you need to do this?

will4ntz, 2025-05-23:
@fettahyildizz Up until that point, there was no release available for the latest version of the jetpack that includes the "supermode" update.