Seeed-Studio/Linux_for_Tegra #35: fatal error missing dts file for j401

mrKallah, 2025-04-08:
I have tried to follow the seed install instructions on two machines now but I get the following error on step 7 during the ./nvbuild.sh. I have tried multiple times, one of the installs are fairly fresh. 

```
/home/username/seed/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/../..//hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j401-p3768-0000+p3767-0000-recomputer.dts:6:10: fatal error: tegra234-j401-p3768-0000+p3767-recomputer-common.dts: No such file or directory
    6 | #include "tegra234-j401-p3768-0000+p3767-recomputer-common.dts"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/username/seed/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/Makefile:118: /home/username/seed/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts//hardware/nvidia/t23x/nv-public/nv-platform/tegra234-j401-p3768-0000+p3767-0000-recomputer.dtb] Error 1
make[1]: Leaving directory '/home/username/seed/Linux_for_Tegra/source/kernel_out'
make: *** [Makefile:173: nvidia-dtbs] Error 2
make: Leaving directory '/home/username/seed/Linux_for_Tegra/source/kernel_out'
```

Listing the files in the offending folder using:

`ls /home/username/seed/Linux_for_Tegra/source/kernel_out/kernel-devicetree/generic-dts/../..//hardware/nvidia/t23x/nv-public/nv-platform/`

returns
```

Makefile                                                    tegra234-j501x-0000+p3701-0004-reserver.dts
tegra234-camera-p3785.dtsi                                  tegra234-j501x-0000+p3701-0004-reserver-gmsl.dts
tegra234-dcb-j501x-0000-p3701-0000.dtsi                     tegra234-j501x-0000+p3701-0005-reserver.dts
tegra234-dcb-p3737-0000-p3701-0000.dtsi                     tegra234-j501x-0000+p3701-0005-reserver-gmsl.dts
tegra234-j201-p3768-0000+p3767-0000-recomputer-indu.dts     tegra234-p3701-0000.dtsi
tegra234-j201-p3768-0000+p3767-0001-recomputer-indu.dts     tegra234-p3701-0000-prod-overlay.dtsi
tegra234-j201-p3768-0000+p3767-0003-recomputer-indu.dts     tegra234-p3701-0005.dtsi
tegra234-j201-p3768-0000+p3767-0004-recomputer-indu.dts     tegra234-p3701-0008.dtsi
tegra234-j201-p3768-0000+p3767-recomputer-indu-common.dts   tegra234-p3737-0000.dtsi
tegra234-j401-p3768-0000+p3767-0000-recomputer.dts          tegra234-p3737-0000+p3701-0000-nv.dts
tegra234-j401-p3768-0000+p3767-0000-recomputer-robo.dts     tegra234-p3737-0000+p3701-0004-nv.dts
tegra234-j401-p3768-0000+p3767-0000-recomputer-super.dts    tegra234-p3737-0000+p3701-0005-nv.dts
tegra234-j401-p3768-0000+p3767-0000-reserver-indu.dts       tegra234-p3737-0000+p3701-0008-nv.dts
tegra234-j401-p3768-0000+p3767-0001-recomputer.dts          tegra234-p3737-0000+p3701-xxxx-nv-common.dtsi
tegra234-j401-p3768-0000+p3767-0001-recomputer-robo.dts     tegra234-p3740-0002.dtsi
tegra234-j401-p3768-0000+p3767-0001-recomputer-super.dts    tegra234-p3740-0002+p3701-0008-nv-common.dtsi
tegra234-j401-p3768-0000+p3767-0001-reserver-indu.dts       tegra234-p3740-0002+p3701-0008-nv.dts
tegra234-j401-p3768-0000+p3767-0003-recomputer.dts          tegra234-p3740-0002+p3701-0008-nv-safety.dts
tegra234-j401-p3768-0000+p3767-0003-recomputer-robo.dts     tegra234-p3740-0002+p3701-0008-safety.dtsi
tegra234-j401-p3768-0000+p3767-0003-recomputer-super.dts    tegra234-p3767-0000.dtsi
tegra234-j401-p3768-0000+p3767-0003-reserver-indu.dts       tegra234-p3768-0000.dtsi
tegra234-j401-p3768-0000+p3767-0004-recomputer.dts          tegra234-p3768-0000+p3767-0000-nv.dts
tegra234-j401-p3768-0000+p3767-0004-recomputer-robo.dts     tegra234-p3768-0000+p3767-0000-nv-px1.dts
tegra234-j401-p3768-0000+p3767-0004-recomputer-super.dts    tegra234-p3768-0000+p3767-0000-nv-super.dts
tegra234-j401-p3768-0000+p3767-0004-reserver-indu.dts       tegra234-p3768-0000+p3767-0000-nv-taylor-high.dts
tegra234-j401-p3768-0000+p3767-recomputer-robo-common.dts   tegra234-p3768-0000+p3767-0000-nv-taylor-low.dts
tegra234-j401-p3768-0000+p3767-recomputer-super-common.dts  tegra234-p3768-0000+p3767-0001-nv.dts
tegra234-j401-p3768-0000+p3767-reserver-indu-common.dts     tegra234-p3768-0000+p3767-0001-nv-super.dts
tegra234-j40mini-p3768-0000+p3767-0000-recomputer.dts       tegra234-p3768-0000+p3767-0003-nv.dts
tegra234-j40mini-p3768-0000+p3767-0001-recomputer.dts       tegra234-p3768-0000+p3767-0003-nv-super.dts
tegra234-j40mini-p3768-0000+p3767-0003-recomputer.dts       tegra234-p3768-0000+p3767-0004-nv.dts
tegra234-j40mini-p3768-0000+p3767-0004-recomputer.dts       tegra234-p3768-0000+p3767-0004-nv-super.dts
tegra234-j40mini-p3768-0000+p3767-recomputer-common.dts     tegra234-p3768-0000+p3767-0005-nv.dts
tegra234-j501x-0000+p3701-0000-reserver.dts                 tegra234-p3768-0000+p3767-0005-nv-super.dts
tegra234-j501x-0000+p3701-0000-reserver-gmsl.dts            tegra234-p3768-0000+p3767-xxxx-nv-common.dtsi
```

Not sure what is going wrong here, but I cannot get past this step

mrKallah, 2025-04-09:
Tested with version 36.4.0 and that worked no problem, so I would assume the issue is only with the latest 36.4.3

ackPeng, 2025-04-09:
@mrKallah  Indeed, two files are missing. I will add the new files after the testing is completed

closed 2025-04-11 by mrKallah