Seeed-Studio/Linux_for_Tegra #40: Tested on 36.4.4?

gstangel, 2025-07-22:
NVIDIA released r36.4.4, which is labeled "Production Quality".  It appears to be minor with no breaking changes, I expect it will be fine with the 36.4.3 kernel patches here.  Has 36.4.4 kernel been tested internally on Seeed boards?

baorepo, 2025-07-22:
@gstangel  Yes, we are currently conducting internal development testing and expect to release it soon. 

@ackPeng It's time to speed up. Everyone has started to pay attention.

gstangel, 2025-07-22:
@baorepo 

great, thanks.  We are only using the J401 carrier and not using any other Seeed components or features.  Is it reasonable to assume that 36.4.3 -> 36.4.4 did not break any core BSP functionality?  

In other words, if we are just using the J401, did 36.4.4 introduce breaking changes to Seeed's latest patches here?

ackPeng, 2025-07-22:
@gstangel There will be no major changes

closed 2025-09-02 by baorepo