Seeed-Studio/Linux_for_Tegra #43: Problem 2 csi cameras imx219

HuysArthur, 2025-08-28:
I recently flashed a J401_V1.0 board with a `Jetson Orin Nano 8GB` using the standard Jetson Linux image (R36.4.4) from https://developer.nvidia.com/embedded/jetson-linux. When I tested two IMX219 cameras connected to the CSI ports, only the camera on the CAM0 (J9) port worked. I discovered that this issue stems from an incompatibility between NVIDIA's original devboard and the J401_V1.0 board. To address this, I applied the tegra234-p3767-camera-p3768-imx219-dual-seeed.dtbo overlay provided in this repository using the jetson-io.py script and rebooted the system. However, the issue persists, with only the CAM0 port camera functioning. Am I missing a step or configuration to enable both cameras?

ackPeng, 2025-09-01:
@HuysArthur This is due to a mipi configuration issue. The two boards use different mini interfaces. You can use this dtbo file:https://github.com/Seeed-Studio/Linux_for_Tegra/blob/r36.4.3/source/hardware/nvidia/t23x/nv-public/overlay/tegra234-p3767-camera-p3768-imx219-dual-seeed.dts

ackPeng, 2025-09-08:
@HuysArthur Is it working?

ackPeng, 2025-09-22:
This isuue is about to be closed. Please open a new one if you have new questions.

closed 2025-09-22 by ackPeng