Seeed-Studio/Linux_for_Tegra #44: Failed to build r35.6.1 for Xavier NX

bessszilard, 2025-09-03:
Hi Seeed studio team,

I failed to build the L4T r35.6.1 for Xavier NX. On the `r35.6.1` branch. The link given in the README `wget https://developer.nvidia.com/downloads/embedded/l4t/r35_release_v5.0/release/tegra_linux_sample-root-filesystem_r35.6.1_aarch64.tbz2` link doesn't work, I got `ERROR 404: Not Found`.
On Nvidia Official page I found the link for the [r32 6.1 sample root filesystem](https://developer.nvidia.com/downloads/embedded/l4t/r35_release_v6.1/tegra_linux_sample-root-filesystem_r35.6.1_aarch64.tbz2), what I downloaded and extracted. 

When I proceeds with the steps, I got stuck when I call the `./nvbuild.sh -i -r ${INSTALL_MOD_PATH}` I got error:

```
./nvbuild.sh: line 10: /home/szilard/seeed/xxx/Linux_for_Tegra/source/nvcommon_build.sh: No such file or directory
```

If you have a built image for the Xavier NX `r35.6.1`, can you share it? On the [Seeed studio's wiki page,](https://wiki.seeedstudio.com/reComputer_Industrial_Getting_Started/) I only saw the 5.1.3 which is r35.5.

Thanks,
Szilard B

ackPeng, 2025-09-04:
@bessszilard There is indeed a problem with the original branch. I have reconstructed this branch and it has passed the compilation test locally. Please re-download this branch and recompile it.

ackPeng, 2025-09-04:
@bessszilard We are planning to conduct rigorous factory testing on this version of Jectpack, but it is not yet completed, so it may have some problems. If you encounter any problems, please feel free to discuss them and we will solve them.

bessszilard, 2025-09-05:
> @bessszilard There is indeed a problem with the original branch. I have reconstructed this branch and it has passed the compilation test locally. Please re-download this branch and recompile it.

@ackPeng , I did a clean build by deleting all the files and redoing all the steps. I noticed that you updated the link for the NVIDIA filesystem with wget, which is great. However, I’m still getting the same error.

```
./nvbuild.sh -i -r ${INSTALL_MOD_PATH}
Building kernel-5.10 sources
make: Entering directory '/home/szilard/seeed/xxx/Linux_for_Tegra/source/kernel/kernel-5.10'
arch/arm64/configs/tegra_defconfig:1406:warning: override: reassigning to symbol TCG_TPM
#
# No change to .config
#
make: Leaving directory '/home/szilard/seeed/xxx/Linux_for_Tegra/source/kernel/kernel-5.10'
  CALL    scripts/atomic/check-atomics.sh
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
/home/szilard/seeed/xxx/Linux_for_Tegra/source/kernel/nvidia/drivers/net/ethernet/nvidia/nvethernet/Makefile:60: /home/szilard/seeed/xxx/Linux_for_Tegra/source/kernel/nvidia/drivers/net/ethernet/nvidia/nvethernet/nvethernetrm/include/config.tmk: No such file or directory
make[4]: *** No rule to make target '/home/szilard/seeed/xxx/Linux_for_Tegra/source/kernel/nvidia/drivers/net/ethernet/nvidia/nvethernet/nvethernetrm/include/config.tmk'.  Stop.
make[3]: *** [scripts/Makefile.build:504: drivers/net/ethernet/nvidia/nvethernet] Error 2
make[2]: *** [scripts/Makefile.build:504: drivers/net/ethernet] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [scripts/Makefile.build:504: drivers/net] Error 2
make: *** [Makefile:1864: drivers] Error 2
make: *** Waiting for unfinished jobs....
```

> @bessszilard We are planning to conduct rigorous factory testing on this version of Jectpack, but it is not yet completed, so it may have some problems. If you encounter any problems, please feel free to discuss them and we will solve them.

Do you know how long it will take? If you’ve already built it for Xavier NX, could you upload the file so I can test it? I just want to see whether the distro update will improve our product quality.

bessszilard, 2025-09-08:
@ackPeng yes, I successfully compiled the `r35.6.1`. Thank for your help 🙂

closed 2025-09-08 by bessszilard